arduino-libraries/MKRGSM #8: Problem with GsmWebServer example

garciavazquez, 2018-01-17:
The server is deployed correctly, and if I make a request from my terminal with wget it is answered. But only the first request. After closing the connection with the client, the serial monitor of Arduino is showing the following message in loop, and it doesn't answer more requests:

AT+USORD=1,0
ERROR

sandeepmistry, 2018-01-22:
Hi @garciavazquez,

Thanks for the report, could you please try the changes from pull request #9 and provide your feedback.

FrancMunoz, 2018-01-24:
Hi!

I think the problem happens when connection is closed localy by GSMClient.stop() this call closes the socket (sents AT+USOCL but does not update _childSockets[i].socket to -1; in GSMServer so when call GSMServer.available() we get an infinite bucle. I didn't tried pull request #9.

I've solved adding a function in Modem.cpp that calls UrcHadlers:

```
void ModemClass::callUrcHandlers(String buffer)
{
	for (int i = 0; i < MAX_URC_HANDLERS; i++) {
    	if (_urcHandlers[i] != NULL) {
			_urcHandlers[i]->handleUrc(buffer);
		}
	}
}
```

Then in GSMClient::stop under MODEM.waitForResponse added:

```
String cmd="+UUSOCL: ";
cmd.concat(_socket);
MODEM.callUrcHandlers(cmd);
_socket = -1;
```

That emulates closing socket AT received by the modem and correctly handles server active sockets. 

Attached are the changed files (sorry can't make a pull request right now).

[MKRChanges.zip](https://github.com/arduino-libraries/MKRGSM/files/1658725/MKRChanges.zip)

garciavazquez, 2018-01-24:
Hi!  @sandeepmistry 
It seems to work properly
Thanks!

sandeepmistry, 2018-01-24:
@garciavazquez great, thanks for trying and the feedback! We'll go ahead and merge, then do another new library release.

@FrancMunoz thanks for the info, it fixes the behaviour you described, let's go for PR #9 for now.

sandeepmistry, 2018-01-24:
Fixed via #9.

A new version 1.1.1 of the library will be available in the library manager shortly.

closed 2018-01-24 by sandeepmistry