arduino-libraries/MKRGSM #16: Externalize URCs handler with a callback

FrancMunoz, 2018-03-10:
Hi! 

I'd like to make a suggestion. I've been working in a large project and need to know for example when a SMS is received instantly without have to pooling if sms is available (that is very slow). So I've created a callback in `GSMClient` to receive all URCs in my code and parse if necessary. 

So in `GSMClient.h` added:

`  void (*urcCallback)(const String& urc);`

and in GSMClient.cpp added the call:

```
void GSMClient::handleUrc(const String& urc)
{
  if(urcCallback!=NULL) urcCallback(urc);
  if (urc.startsWith("+UUSOCL: ")) {
    int socket = urc.charAt(urc.length() - 1) - '0';

    if (socket == _socket) {
      // this socket closed
      _socket = -1;
    }
  } else if (urc.startsWith("+UUSORD: ") && urc.endsWith(",4294967295")) {
    // SSL disconnect
    int socket = urc.charAt(9) - '0';

    if (socket == _socket) {
      // this socket closed
      _socket = -1;
    }
  }
}
```

So in code could receive SMS notifications. The only issue is the use of flags during the callback to be  executed at main loop, because if an operation is done in the callback all seems to fail due the design of the modem code (that I think is nice). 

So:

```
bool isSMS=false;

void handleUrc(const String& urc) {
    Serial.print("URC: ");
    Serial.println(urc);
    if(urc.startsWith("+CMTI")) {
        isSMS=true;
    }
}

void setup() {
    // Setup GSM... 
    client.urcCallback=handleUrc;
    MODEM.send("AT+CNMI=2,1");  // Enables URC SMS
}

void loop() {
    if(isSMS) {
        isSMS=false;
        //checkSMS();
    }
}
```

This will give more power and control to final coding.

What do you think?

sandeepmistry, 2018-03-20:
@FrancMunoz thank you for proposing this. At this time I would prefer to leave this out.

FrancMunoz, 2018-03-21:
Hi @sandeepmistry! 
I see it could be complicated and could generate some problems but improves control on final coding. I would use `ModemClass::addUrcHandler` because I would like to use your future improvements.

Thank you!

closed 2018-09-12 by sandeepmistry