arduino-libraries/MKRGSM #17: Overload debug() in Modem.cpp

FrancMunoz, 2018-03-10:
Hi!

I've another suggestion. Is it possible to overload ModemClass::debug with ModemClass::debug(bool state)?

```
void ModemClass::debug(bool state)
{
  _debug = state;
}
```

The reason (as it's evidenced) is to be capable to enable / disable debug at runtime. So in main loop will be possible to enable or disable from monitor, event sending AT commands to SARA from it:

This is an example that I use to enable debug when something goes wrong:

```
void loop() {
    char c;

    if(Serial.available()) {
        c=Serial.read();
        if(c=='#') {
            if(MODEM._debug) {
                MODEM._debug=false;
                Serial.println("Disable DEBUG");
            }
        } else if(c!='\n' && c!='\r') {
            if(!MODEM._debug) {
                MODEM._debug=true;            
                Serial.println("Enable DEBUG");
            }
        }
        MODEM.write(c);
    }
}
```

I think it's a simple change that would give us more flexibility to debug. I used to debug GSM status as @alvarolb asked in #15, and then I'm able to send AT commands to restore GSM or GPRS status so final coding is more stable and good.

What do you think?

Tahnk you!

alvarolb, 2018-03-12:
Hi @FrancMunoz, I think it could be helpful for debugging. What did you find about GSM/GPRS status? What is the way you currently monitor its state? Do we need to restart GSM/GPRS on cell changes? Signal loss? Thanks.

sandeepmistry, 2018-03-20:
What about having `MODEM.debug()` and `MODEM.noDebug()`?

FrancMunoz, 2018-03-21:
That would be nice.

Rocketct, 2018-04-24:
Hi everyone, @sandeepmistry  I have tried to enable and disable the debug through your function and it works fine.

sandeepmistry, 2018-04-24:
Closed via PR #20.

closed 2018-04-24 by sandeepmistry