arduino-libraries/MKRGSM #21: Loop for ever at CLIENT_STATE_WAIT_CLOSE_SOCKET when server is unavailable (GSMSSLClient)

w3p706, 2018-04-06:
If I try to connect to a Server which is not available, the connect function will loop for ever after the following sequence:

    AT+USOCR=6
    +USOCR: 0
    
    OK
    AT+USOSEC=0,1
    OK
    AT+USOCO=0,"valid.domain.port.closed.net",8883
    ERROR
    AT+USOCL=0
    +CME ERROR: operation not allowed
    
    +UUSOCL: 0

Another Variant ist this:

    AT+USOCR=6
    +USOCR: 0

    OK
    AT+USOSEC=0,1
    OK
    AT+USOCO=0,"valid.domain.port.closed.net",8883
    ERROR

    +UUSOCL: 0
    AT+USOCL=-1
    +CME ERROR: operation not supported

    +UUPSDD: 0

The `_state` hangs at CLIENT_STATE_WAIT_CLOSE_SOCKET

The state machine never enters `case CLIENT_STATE_WAIT_CLOSE_SOCKET` the URC for UUSOCL is not run.

If I change my code like this, it works:

    case CLIENT_STATE_WAIT_CONNECT_RESPONSE: {
      if (ready > 1) {
        _socket = -1;
      }
      _state = CLIENT_STATE_IDLE;
      break;
    }

And most probably the same has to be done for     

    case CLIENT_STATE_WAIT_ENABLE_SSL_RESPONSE: {
      if (ready > 1) {
        _state = CLIENT_STATE_IDLE;
        break;
      } else {
        _state = CLIENT_STATE_CONNECT;
      }

      ready = 0;
      break;
    }

sandeepmistry, 2018-04-06:
Hi @w3p706,

Could you please provide a minimal sketch to reproduce this issue, I tried with the [GsmWebClient.ino](https://github.com/arduino-libraries/MKRGSM/blob/master/examples/GsmWebClient/GsmWebClient.ino) and changing the server + port to `valid.domain.port.closed.net` and `8883`(with both GSMClient and GSLSSLClient) - but things seemed ok. Here's the debug output I get:

```
connecting...
AT+USOCR=6

+USOCR: 0

OK
AT+USOSEC=0,1

OK
AT+USOCO=0,"valid.domain.port.closed.net",8883

ERROR
AT+USOCL=0

ERROR
connected

+UUSOCL: 0

disconnecting.
```

It's confusing from the expected behaviour from: https://github.com/arduino-libraries/MKRGSM/blob/master/src/GSMClient.h#L49-L53

```
  /** Connect to server by IP address
      @param (IPAddress)
      @param (uint16_t)
      @return returns 0 if last command is still executing, 1 success, 2 if there are no resources
   */
```

The `client.connect(...)` returns 2 on failure, instead of 0 as per the Client spec.

w3p706, 2018-04-11:
Hi @sandeepmistry 

I can reproduce the issue with the GsmSSLWebClient example if I insert 

    MODEM.send("AT+CMEE=2");
    (MODEM.waitForResponse(1000) == 1);

after the modem connected (line ~60). If then the remote party is not available (port closed) the system will enter a loop for ever in state `CLIENT_STATE_WAIT_CLOSE_SOCKET` as described above.

`AT+CMEE=2` will report mobile termination error (for error codes see [page 745](https://www.u-blox.com/sites/default/files/u-blox-CEL_ATCommands_%28UBX-13002752%29.pdf))

behavior with `AT+CMEE=2`

    AT+USOCO=0,"DOMAIN",8883
    ERROR
    AT+USOCL=0
    +CME ERROR: operation not allowed
    +UUSOCL: 0

behavior without `AT+CMEE=2`

    AT+USOCO=0,"mqtt.w3p.net",8884
    ERROR
    AT+USOCL=0
    ERROR

This will make the logic in `Modem.cpp:252` fail:
    
    responseResultIndex = _buffer.lastIndexOf("ERROR\r\n");
    if (responseResultIndex != -1) {
        _ready = 2;

Bottom line is that this is quite an edge case, so I closed this issue.

closed 2018-04-11 by w3p706