arduino-libraries/MKRGSM #22: Problem With MQTT MKR1400GSM example

Giorgiofox, 2018-04-09:
Hello, 
I'm using this code but after some second (sometimes 15, sometimes 120) it stop to work, and hang up. I've tried also with a blinking led, but also the led stop flash. Seem that the board is freezeed. 

I'm using i++ as counter to measure seconds in terminal. 
My previous skecth are more complicated, but I've started again using example to found the problem.

Thanks in advance

Giorgio

`#include <MKRGSM.h>
#include <MQTTClient.h>

const char pin[]      = "";
const char apn[]      = "tre.it";
const char login[]    = "";
const char password[] = "";
int i;
GSMClient net;
GPRS gprs;
GSM gsmAccess;
MQTTClient client;

unsigned long lastMillis = 0;

void setup() {
  Serial.begin(115200);

  // Note: Local domain names (e.g. "Computer.local" on OSX) are not supported by Arduino.
  // You need to set the IP address directly.
  client.begin("myservermqtt.com", 1883, net);
  client.onMessage(messageReceived);

  connect();
}

void connect() {
  // connection state
  boolean connected = false;

  Serial.print("connecting to cellular network ...");

  // After starting the modem with gsmAccess.begin()
  // attach to the GPRS network with the APN, login and password
  while (!connected) {
    if ((gsmAccess.begin(pin) == GSM_READY) &&
        (gprs.attachGPRS(apn, login, password) == GPRS_READY)) {
      connected = true;
    } else {
      Serial.print(".");
      delay(1000);
    }
  }
  
  Serial.print("\nconnecting...");
  while (!client.connect("arduino", "myusername", "mypassword")) {
    Serial.print(".");
    delay(1000);
  }

  Serial.println("\nconnected!");

  client.subscribe("MKR1400/COMMAND/#");
  // client.unsubscribe("/hello");
}

void loop() {
  client.loop();

  if (!client.connected()) {
    connect();
  }

  // publish a message roughly every second.
  if (millis() - lastMillis > 1000) {
    lastMillis = millis();
    i++;
    
    client.publish("MKR1400/POSITION/", "{Position: position test}");
    Serial.println(i);
    
    
  }
}

void messageReceived(String &topic, String &payload) {
  Serial.println("incoming: " + topic + " - " + payload);`

w3p706, 2018-04-11:
Hi Giorgio

As I'm too experimenting with the MKRGSM and MQTT at the moment I tried your code.
Your code works nice with `broker.shiftr.io`. I Stopped it at i=775.

Your code structure also works with SSL with my mosquitto broker.

But your code fails, if i restart the Mosquitto broker, and then I have a lot of the following entries in the brokers log: 

    1523465876: New connection from xx.xxx.xx.xxx on port 8883.
    1523465892: Socket error on client <unknown>, disconnecting.

After approx. 8 minutes, the client could connect again to the broker. 

You could 
- try another broker
- watch the log of your broker
- I had some success initially with the following options:
        `_mqtt.setOptions(60, false, 2500);`
- I also had success with implementing a watchdog (`Adafruit_SleepyDog`), but this is an ugly hack.
- enable gsm debug output `GSM gsmAccess(true)'; then you see more whats going on.

Also keep an eye on my other open issue (https://github.com/arduino-libraries/MKRGSM/issues/21), as this could be related.

Rocketct, 2018-05-03:
Hi all, @Giorgiofox i have test your skecth, on my GSM for two days without issue, as MQTT server i have used the test server of mosquitto, have you tried to use another broker?

stefanidisgeorge, 2018-05-07:
have you tried cloudmqtt?

Giorgiofox, 2018-05-16:
Sorry for delay, I will do some test today.

Rocketct, 2018-05-16:
Ok thanks @Giorgiofox , @stefanidisgeorge yes i have tried also with cloudmqtt, with this sketch, for 5 hours without problems.

Giorgiofox, 2018-05-16:
Uhm: 
I have both, an account on CloudMQTT and my own MQTT broker. 
First test on CloudMQTT hang on 81
First test on my own Mosquitto broker hang on 76

After this I have enabled GSM debug mode, and this is the result (only last rows after last 'I') https://pastebin.com/km32dCpL

Maybe is my board with some problem?
Or my SIM?

Giorgiofox, 2018-05-16:
I have tried another Data SIM, same problem: hang at 21

Rocketct, 2018-05-16:
@Giorgiofox ok ok one question, which are the version of the libraries that you use?

Giorgiofox, 2018-05-16:
I'm using MKRGSM 1.1.1, and as I can see there are available version 1.1.2, let me try this new version

Giorgiofox, 2018-05-16:
with 1.1.2 version of library the problem seem solved!