arduino-libraries/MKRGSM #23: SSL/TLS security

daz, 2018-04-11:
I've added an API for importing certificates, with an example that connects to AWS IoT which uses TLS 1.2.

It currently works like this:

1. Create a `GSMSecurity` profile
2. Import certificates to the profile
3. Configure SSL/TLS settings for the profile
4. Apply the profile to the `GSMSSLClient` client

```cpp
  GSMSecurity profile;

  profile.setRootCertificate(SECRET_ROOT_CERT);
  profile.setClientCertificate(SECRET_CLIENT_CERT);
  profile.setPrivateKey(SECRET_PRIVATE_KEY);
  profile.setValidation(SSL_VALIDATION_ROOT_CERT);
  profile.setVersion(SSL_VERSION_TLS_1_2);
  profile.setCipher(SSL_CIPHER_AUTO);
  gsmClient.setSecurityProfile(profile);
```

I opted for the profile approach rather than add methods to `GSMClient` because the modem can only hold 9 certificates in total, and they persist between boots. Importing certs after 9 will fail, so we need a way to list and remove them, which imo would bloat the `GSMClient` class.

Here's my fork, I didn't wanna dive in and create a pull request in case there's a better way to tackle this

https://github.com/daz/MKRGSM

facchinm, 2018-04-11:
Hi @daz ,
that's great! Thank you so much for diving into this.
I like the approach a lot since it lets the user upload the certificates in the usual way but it also allows us to make a wrapper sketch that interacts with an external loader (like WiFi101 one https://github.com/arduino-libraries/WiFi101-FirmwareUpdater-Plugin).
Feel free to propose a PR when ready or also earlier if you need some feedback from the community!

daz, 2018-04-12:
Added PR #24

closed 2018-04-12 by daz