arduino-libraries/MKRGSM #25: MKRGSM does not work with ArduinoHttpClient (or the other way around)

lathoub, 2018-04-18:
I have tried to call GET and POST using the [ArduinoHttpClient](https://github.com/arduino-libraries/ArduinoHttpClient) library, but unfortunately GET works only once and POST times-out.

Not sure where the issue is, with this library or with ArduinoHttpClient.

Great to see the usage of the GSMClient that inherits from Client, that simplifies (normally) the usage of HTTP commands.

Code:

Just below `int port = 80;` add:
```
#include <ArduinoHttpClient.h>
HttpClient http(client, server, port);
```

in loop:

For GET:
```
http.get(path);
int statusCode = http.responseStatusCode();
String body = http.responseBody();
```

for POST
```
http.post(path, "application/json", "{}");
int statusCode = http.responseStatusCode();
String body = http.responseBody();
```

Rocketct, 2018-04-23:
Hi @lathoub  starting from your Examples i have done the following modifications in MKRGSM  libraries https://github.com/arduino-libraries/MKRGSM/pull/28 and ArduinoHttpclient librarries https://github.com/arduino-libraries/ArduinoHttpClient/pull/41 this ones reduce the response time of get and post functions.
Can you try this code and see if it work as you expect for your examples? about the get function in my test, it work fine and print in serial monitor the results 6 times.
Moreover for get function i have to rewrite the available() function in https://github.com/arduino-libraries/MKRGSM/pull/28 because with  a great buffer length your example with httpbin.org require 33 seconds instead of  1 second.

sandeepmistry, 2018-09-12:
Hi @lathoub,

Would you be able to try again with the changes in pull request #44?

sandeepmistry, 2018-10-03:
With PR #44 and #54, ArduinoHTTPClient is working with this library for both TCP and SSL connections for me.

Rocketct, 2018-10-10:
Hi all @sandeepmistry @lathoub  tested  https://github.com/arduino-libraries/MKRGSM/pull/26 wit the PR https://github.com/arduino-libraries/MKRGSM/pull/44  and https://github.com/arduino-libraries/MKRGSM/pull/54 it also work for me

sandeepmistry, 2018-10-18:
Closed by #44 and #54.

closed 2018-10-18 by sandeepmistry