arduino-libraries/MKRGSM #27: MKR1400 hangs when running the GsmWebClient sketch

lathoub, 2018-04-22:
My MKRGSM1400 hangs after a random amount of time (sometime after a couple of seconds, sometimes after an hour).

The Code (based on the `GsmWebClient` example, but then 'get' is put in a loop). The` Serial Monitor` provides a trace (see below)

If I want to upload new code to the MKR1400, I need to reset it first using the little reset button to 'unhang' it.

I'm using a MacBook Pro (15-inch, 2017) with High Sierra (10.13.4 (17E199)), Thunderbolt 3 to USB dongle. The MKR1400 has a charged 850mAh battery attached to it (I also tried a 2000mAh battery, same result).

```
#include <MKRGSM.h>

#include "arduino_secrets.h"
// Please enter your sensitive data in the Secret tab or arduino_secrets.h
// PIN Number
const char PINNUMBER[]     = SECRET_PINNUMBER;
// APN data
const char GPRS_APN[]      = SECRET_GPRS_APN;
const char GPRS_LOGIN[]    = SECRET_GPRS_LOGIN;
const char GPRS_PASSWORD[] = SECRET_GPRS_PASSWORD;

// initialize the library instance
GSMClient client;
GPRS gprs;
GSM gsmAccess(true);

// URL, path and port (for example: arduino.cc)
char server[] = "arduino.cc";
char path[] = "/asciilogo.txt";
int port = 80; // port 80 is the default for HTTP

void setup() {
  // initialize serial communications and wait for port to open:
  Serial.begin(9600);
  while (!Serial) {}

  Serial.println("Starting Arduino web client.");
  // connection state
  boolean connected = false;

  // After starting the modem with GSM.begin()
  // attach the shield to the GPRS network with the APN, login and password
  while (!connected) {
    if ((gsmAccess.begin(PINNUMBER) == GSM_READY) &&
        (gprs.attachGPRS(GPRS_APN, GPRS_LOGIN, GPRS_PASSWORD) == GPRS_READY)) {
      connected = true;
    } else {
      Serial.println("Not connected");
      delay(1000);
    }
  }
}

void loop() {

  Serial.println("connecting...");

  // if you get a connection, report back via serial:
  if (client.connect(server, port))
  {
    Serial.println("connected");
    // Make a HTTP request:
    client.print("GET ");
    client.print(path);
    client.println(" HTTP/1.1");
    client.print("Host: ");
    client.println(server);
    client.println("Connection: close");
    client.println();

    Serial.println("awaiting response");

    while (true) {
      // if there are incoming bytes available
      // from the server, read them and print them:
      if (client.available())
        char c = client.read();

      // if the server's disconnected, stop the client:
      if (!client.available() && !client.connected()) {
        Serial.println();
        Serial.println("disconnecting.");
        client.stop();
        break;
      }
    }
  }
  else
  {
    // if you didn't get a connection to the server:
    Serial.println("connection failed");
  }

  delay(5000);
}
```

The content of the serial monitor:

```
21:49:19.604 -> Starting Arduino web client.
21:49:23.393 -> ⸮AT

21:49:23.605 -> OK
21:49:23.605 -> AT+IPR=921600

21:49:23.605 -> OK
21:49:23.712 -> AT

21:49:23.712 -> OK
21:49:23.712 -> AT+UPSV=3

21:49:23.712 -> OK
21:49:23.817 -> AT+CPIN?
```

...

```
21:51:57.115 -> +UUSORD: 0,2
21:51:57.115 -> AT+USORD=0,1

21:51:57.115 -> +USORD: 0,1,"20"
21:51:57.115 -> 
21:51:57.115 -> OK
21:51:57.115 -> 
21:51:57.115 -> +UUSORD: 0,1
21:51:57.115 -> AT+USORD=0,1

21:51:57.115 -> +USORD: 0,1,"0A"
21:51:57.115 -> 
21:51:57.115 -> OK
21:51:57.369 -> 
21:51:57.369 -> +UUSOCL: 0
21:51:57.369 -> 
21:51:57.369 -> disconnecting.
21:52:02.352 -> connecting...
21:52:02.463 -> AT+USOCR=6

21:52:02.463 -> +USOCR: 0
21:52:02.463 -> 
21:52:02.463 -> OK
21:52:02.675 -> AT+USOCO=0,"arduino.cc",80

21:52:02.961 -> OK
21:52:02.961 -> connected
21:52:02.961 -> AT+USOWR=0,4,"47455420"

21:52:02.961 -> +USOWR: 0,4
21:52:02.961 -> 
21:52:02.961 -> OK
21:52:02.961 -> AT+USOWR=0,14,"2F61736369696C6F676F2E747874"

21:52:02.961 -> +USOWR: 0,14
21:52:02.961 -> 
21:52:02.961 -> OK
21:52:02.961 -> AT+USOWR=0,9,"20485454502F312E31"

21:52:02.997 -> +USOWR: 0,9
21:52:02.997 -> 
21:52:02.997 -> OK
21:52:02.997 -> AT+USOWR=0,2,"0D0A"

21:52:02.997 -> +USOWR: 0,2
21:52:02.997 -> 
21:52:02.997 -> OK
21:53:05.524 -> 
21:53:05.524 -> +UUSOCL: 0
```

MKR1400 hangs at 21:53:05.524, not finishing the +UUSOCL command
(at other tries, the code hangs elsewhere - so not always at the same point)

lathoub, 2018-04-23:
Here the MKR1400 stalled during reading of the response:

```
+UUSORD: 0,241
AT+USORD=0,1
+USORD: 0,1,"74"
OK
+UUSORD: 0,240
AT+USORD=0,1
+USORD: 0,1,"65"
OK
+UUSORD: 0,239
AT+USORD=0,1
+USORD: 0,1,"6E"
OK
+UUSORD: 0,238
AT+USORD=0,1
+USORD: 0,1,"74"
OK
+UUSORD: 0,237
```

lathoub, 2018-04-24:
I bought another brand new MKR1400 from the Arduino store, hoping the above hanging just due to hardware failure - nope! New unit hangs as well..... 

What is the ideal power setup?
Am I the only one experiencing hangs/freezes?

Here is the log:
```
OK

+UUSORD: 0,119
AT+USORD=0,1

+USORD: 0,1,"67"

OK

+UUSORD: 0,118
AT+USORD=0,1

+USORD: 0,1,"74"

OK

+UUSORD: 0,117
AT+USORD=0,1

+USORD: 0,1,"68"

OK

+UUSORD: 0,116
AT+USORD=0,1

+USORD: 0,1,"3A"

OK

+UUSORD: 0,115
```

FrancMunoz, 2018-04-25:
Hi @lathoub , Please take a look at #10, I think it could help you. All in all update your libraries to last version.

Hope it helps!

lathoub, 2018-04-25:
Thanks @FrancMunoz 

When reading #10, I understand that the issue is still under investigation (with some pointers how to solve the issue). Let me try to include your alternatives.

Not sure if I understand the 'update my libraries'. In the sketch above I don't use any libraries at all (except for MKRGSM.h)

FrancMunoz, 2018-04-25:
Hi @lathoub , with 'update your libraries' I mean make sure you use last version of MKRGSM and also **SAMD core**. You can update both with arduino IDE first under **Libraries Manager**, second under **Boards Manager**. More information about SAMD core:

[Arduino SAMD Core](https://github.com/arduino/ArduinoCore-samd)

lathoub, 2018-04-29:
I got [TinyGSM](https://github.com/vshymanskyy/TinyGSM) to work with the MKR GSM and is running stable for 3 hours now (doing HTTP GET) - I'll leave it overnight and see it if freezes or not.

lathoub, 2018-04-30:
Pull request to the example: https://github.com/vshymanskyy/TinyGSM/pull/161

Prof62, 2018-05-28:
+1 on an urgent fix for this.

I'm using the current release version of the SAMD core (1.6.18) and the MKRGSM library (1.1.2) but still experience random hangs after anywhere between a few seconds and a few hours.
 
Issue #10 is closed so presumably there is no official active investigation going on there (Although there are a few recent requests to reopen it) though it has been merged with issue #14 which is still open but relates to a slightly different issue and hasn't had any activity since the merge point in February.

The buffering approach described by @FrancMunoz in issue #10 does not resolve the problem but his delay solution is looking promising so far. Though, given the random nature of the problem it's going to take a few more days of testing to be certain.

Until this is fixed the MKR GSM 1400 is virtually useless to me as an IOT device which is a pity because it has great potential.

FrancMunoz, 2018-05-28:
Hi! I've found the cause of the infinite loops (pointing to hangs). The `GSMClient::connect()` function aren't timed out so if something goes wrong (for example some URC like `+CREG: 2` is received while connecting -> that means module is not registered) the loop never ends and that's it... a hang.

So, I've updated `Modem.cpp` to handle 20ms just waiting for the difference between loops, as I already suggested (that is, just subtracting to 20 the main loop time), and also `GSMClient.cpp` adding timeouts to all `while (ready() == 0)` loops, and at this time several days without a hang even disconnecting and reconnecting the antenna... I've to say that the sketch code has to be prepared for it negative connect calls... Until this last change when SARA module was registered no hangs ocurred.

I've attach my testings it would be great If you could test and post results... to improve.

[MKRGSM.zip](https://github.com/arduino-libraries/MKRGSM/files/2045960/MKRGSM.zip)

I would like to collaborate with this libraries making PRs, but I guess my thinkings are on other way and I'm a litle bit affraig of making it.

All in all @Prof62, MKR is a nice board, we only need to improve the code and make it stable, I'm totally agree with you. Arduino team is triying to make it very nice... while we only need it to be stable.

Hope it helps!

lathoub, 2018-05-29:
Hi @FrancMunoz 
I appreciate your positiveness, but to be honest: MKR 1400's are sold under the Arduino brand, so I expect quality and active software development. @Prof62 is right, currently the MKR 1400 is useless 

@agdl you work for Arduino (looking at your profile), what is the roadmap to fix this show-stopper issues?

FrancMunoz, 2018-05-29:
Hi! Yes, I told you: I'm totally agree with @Prof62  and you (I got very frustrated seeing MKR hanging). I'm just trying to solve that problem: you see I'm not working nor I'm part of Arduino team... so my interest is purely to solve that annoying issues and help if it's possible.

I've just made some changes to MKRGSM library:

[MKRGSM.zip](https://github.com/arduino-libraries/MKRGSM/files/2047585/MKRGSM.zip)

This one is able to loose registration to network (deattaching and reattaching the antenna), it doesn't hangs for me. I guess even with this situation somebody will test it.

So, @lathoub or @Prof62 could you please test the library attached to this message, it will help me to know if I'm on the right way?

Thank you!

Rocketct, 2018-05-30:
Hi all, i start to work on this issue, i looking into reproducing the problem.

FrancMunoz, 2018-05-30:
Hi @Rocketct, AFAIK the issue appears randomly and it's due to SARA module does not send any response to MKR, that makes infinite loops (waiting data) in code. The library I attached above, the last one, solve this issue adding timeouts and taking control if something goes wrong (that happens too if you remove antenna while receiving or sending data) ... Changes I made are working since two days without any hang... but does NOT solve the main problem: why SARA does not send nothing.

I my case, it happens randomly, it sometimes when connecting: it dies at `+CREG` URC. But most of cases are when transfering or receiving data.

Hope it helps!

Rocketct, 2018-05-31:
ok @FrancMunoz  thank you, i'll try it, have you used a particular sketch to reproduce the issue?

FrancMunoz, 2018-05-31:
@Rocketct , No simply used the same sketch that was failing changing libraries and running for a long time. The #12 is based on it.

FrancMunoz, 2018-05-31:
@Rocketct I've noticed using the above attached library that timeouts are so low so sometimes when network is slow it ends but no error has occurred yet (but control from arduino is not lost). I've increades timeouts to 10000ms. If you test please increase these values. If you need I can send you modificacions.

FrancMunoz, 2018-05-31:
@Rocketct , no, I mean in:

GSMClient::connect()

```
int GSMClient::connect()
{
  unsigned long start;
  if (_synch) {
    start = millis();
    while (ready() == 0) {
      if(millis()-start>10000) {
      	_state = CLIENT_STATE_ERROR_TIMEOUT;
      	return 0;
      }
    }
  } else if (ready() == 0) {
    return 0;
  }

  _state = CLIENT_STATE_CREATE_SOCKET;

  if (_synch) {
  	start = millis();
    while (ready() == 0) {
      delay(100);
      if(millis()-start>10000) {
      	_state = CLIENT_STATE_ERROR_TIMEOUT;
      	return 0;
      }
    }

    if (_socket == -1) {
      return 2;
    }
  }

  return 1;
}

```

GSMClient::write

```
size_t GSMClient::write(const uint8_t* buf, size_t size)
{
  if (_writeSync) {
  	unsigned long start = millis();
    while (ready() == 0) {
      if(millis()-start>10000) {
      	_state = CLIENT_STATE_ERROR_TIMEOUT;
      	return 0;
      }
    }
  } else if (ready() == 0) {
    return 0;
  }
...

```

and  GSMClient::available

```
int GSMClient::available()
{
  unsigned long start = millis();
  if (_synch) {
    while (ready() == 0) {
      if(millis()-start>10000) {
      	_state = CLIENT_STATE_ERROR_TIMEOUT;
      	return 0;
      }
    }
  } else if (ready() == 0) {
    return 0;
  }

  if (_socket == -1) {
    return 0;
  }

  MODEM.poll();

  return _available;
}
```

All in the modified library I uploaded above. I created the state CLIENT_STATE_ERROR_TIMEOUT to know the state of modem and respond if necessary.

daz, 2018-06-06:
In the meantime, how about using a watchdog?

https://github.com/adafruit/Adafruit_SleepyDog/blob/master/utility/WatchdogSAMD.cpp

Rocketct, 2018-06-06:
@FrancMunoz i have test your library and also for me it hangs generally after one day, i had experienced the freeze in writing operation and only one time in `+CREG`

FrancMunoz, 2018-06-06:
Hi @Rocketct , yes... it freezes at these points I'm working to solve that (I need it make it work to finish our project). All in all I'm guessing the problem is in SAMD core, I'm not sure but have some suspicion about it.

In my case, sometimes all goes crazy all commands sent to SARA returns ERROR or there's no return from SARA, sometimes it simple get in an infinite loop, sometimes fails writing and sometimes `+CREG` also fails. Since I made it work with previous version of SAMD core, will try and post here.

@daz , I think a watchdog is a bad practice here, it won't help. Only will have an arduino rebooting without any control (I had it).

FrancMunoz, 2018-06-06:
Ah! @Rocketct , I think it freezes after a day because SARA gets collapsed and all goes crazy.

FrancMunoz, 2018-06-19:
Hi! I've updated SAMD core with [hourly builds](https://github.com/arduino/ArduinoCore-samd#hourly-builds) and it seems that is more stable. Then noticed SARA hangs the Arduino processor because it does not respond to it and then we get infinite while loops.

That's because my didn't the code don't take notice of SARA state and repeated commands cause SARA to stop working. So changed my code to consider SARA status before trying to do anything else.

Then noticed that Movistar (Spain operator) changes IP or forces to reconnect every 30-36 hours (maybe from DHCP lease timeout. 

So changed my code:

1. To know every time the status of SARA
2. To reconnect sockets in case of disconnection (local or remote)
3. To reattach GPRS in case of IP changed (also can be done sending `AT+UPSDA=0,3` command)
4. To reboot (is the same time reboot to restart GSM) in case of GSM failure.

That worked very fine, now we don't have hangs nor have to reboot. It has been working four days without any reboot (first message sent from arduino is a hello to server and server stores all hellos received)

I still using 20ms delay and downgraded modem speed to 57600.

Will update soon hope with good news.

Rocketct, 2018-06-20:
ok @FrancMunoz  i'll try your version of  core immediately, while i had try to reestablish again the connection with the module and try to manage the delay frome skatch.
Why you have daowngrade the speed to 57600?whit 921600 had find some problems?

FrancMunoz, 2018-06-22:
Hi! I downgrade speed because I discovered is more stable in previous versions of Core, right now when I've problems I downgrade and it seems more stable... and then try to upgrade... I think it's not necessary right now... but when desperation comes around... :)

ardiri, 2018-07-11:
i've come across this myself today and this thread sums up my findings as well..

the sad thing is; the library has been designed to be asynchronous in nature (avoid these infinite loops); but the implementation has been half done.. for example; you can pass a false value to the constructors for GSM and GPRS - but, can you check the status? hell no..

`GSM3_NetworkStatus_t GSM::begin(const char* pin, bool restart, bool synchronous)
{
  if (!MODEM.begin(restart)) {
    _state = ERROR;
  } else {
    _pin = pin;
    _state = IDLE;
    _readyState = READY_STATE_CHECK_SIM;

    if (synchronous) {
      while (ready() == 0) {
        delay(100);
      }
    } else {
      return (GSM3_NetworkStatus_t)0;
    }
  }

  return _state;
}`

how does one get the state of the modem if they were to implement asynchronous calls? they cannot; i had to modify the library to add a method to the objects:

`  int lowPowerMode();
  int noLowPowerMode();
+ GSM3_NetworkStatus_t getState() { return _state; };`

and then i could modify my code to be as follows:

`  dbgUART.print(":: register GSM network:   ");
  do
  {
    gsm.begin(PIN_NUMBER, true, true); // force reset of modem
    ts = millis(); while ((gsm.ready() != 0) && millis() - ts < AT_TIMEOUT);
  } while (gsm.getState() != GSM_READY);
  dbgUART.println("done");`

and put the timeout handling in my sketch; with minimal changes to the library. i'm finding now that my code is handing in the connect(), send() or available() methods - and i'll make more changes similar to what FrancMunoz has done to rectify the problem.

ardiri, 2018-07-11:
just a small update; using the code thats half written - sorted out the client being asynchronous as well

int getState() { return _state; }

and inside the write() function comment out the _writeSync (it has a timeout on it)

**//  if (_writeSync)**

you can then connect using the following code:

    dbgUART.print(":: connecting to server:   ");
    client.connect(server, port);
    ts = millis(); while ((client.ready() == 0) && millis() - ts < HTTP_TIMEOUT) ;
    if (client.getState() == 0)
    {
      dbgUART.println("done");
      
      // make a HTTP request: 
      **client.beginWrite(false);**
      client.print("GET ");   client.print(path);     client.println(" HTTP/1.1");
      client.print("Host: "); client.println(server);
      client.println("Connection: close");
      client.println();
      **client.endWrite(false);**

of course; the other changes that FrancMunoz mentions are still an issue - but, i believe there was some design consideration to have asynchronous manipulation to avoid these horrible infinite loops as part of the design; it just hasn't been completed fully.

Rocketct, 2018-07-13:
Hi @ardiri why you need the value of _state? is not sufficient the value returned by ready()

ardiri, 2018-07-15:
actually; i would prefer to have written it as:

`if (client.getState() == CLIENT_STATE_IDLE)
`

but the constant is defined in the .cpp file; ready could refer to a number of conditions:

        if (c == '\n') {
          int responseResultIndex = _buffer.lastIndexOf("OK\r\n");
          if (responseResultIndex != -1) {
            _ready = 1;
          } else {
            responseResultIndex = _buffer.lastIndexOf("ERROR\r\n");
            if (responseResultIndex != -1) {
              _ready = 2;
            } else {
              responseResultIndex = _buffer.lastIndexOf("NO CARRIER\r\n");
              if (responseResultIndex != -1) {
                _ready = 3;
              }
            }
          }

which is deeply nested within the library itself.. in fact; the code is a little badly written; you could technically deal with checking if ready is 1; but, you would have to have found the code above to do so.

andystroz, 2018-07-17:
@FrancMunoz I am running into the same issues as you. I was wondering if you could provide the code you used that solved the problem of SARA hanging the Arduino processor, causing infinite while loops.

FrancMunoz, 2018-07-18:
Hi @andystroz ,

Just take a look at #30... I changed to persistent connection but I think there are some things there that will help you.

sandeepmistry, 2018-09-12:
Hi @FrancMunoz,

Would you be able to open a pull request for your suggestion in https://github.com/arduino-libraries/MKRGSM/issues/27#issuecomment-393537138 ?

Let us know, if you can't we can look into making the changes you suggested.

sandeepmistry, 2018-09-28:
@Rocketct can you please re-test this with what's in master and PR #52 and #53.

Nels52, 2018-09-30:
Hi All,
I am new to the Arduino world in general and the MKRGSM 1400.  This is my first MKRGSM 1400 project.  I am using GSM, GPRS, and GSMClient to periodically send data to an IOT website.

When using the default synchronous mode of operation for GSM, GPRS, and GSMClient I also experienced hangs as noted in this thread.  I found I was able to eliminate these hangs by specifying asynchronous operation for all 3 of these MKRGSM library class instances and driving the state machines for these classes one step at a time using the GSM::ready(), GPRS::ready(), and GSMClient::ready() methods.

In my Arduino script I embed ready() calls for these class instances within loops which have time limits.  If the time limit is reached before a ready() call returns a non-zero value, the script reinitialzes the modem from the beginning with a GSM::begin(PINNUMBER, true, false) and a GPRS::attachGPRS(GPRS_APN, GPRS_LOGIN, GPRS_PASSWORD, false).  This works fine with the GSM and GPRS class instances because the GSM::begin() method performs a hard reset on the modem and sets the GSM _readyState to READY_STATE_CHECK_SIM and the GPRS::attachGPRS() method sets the GPRS _state to GPRS_STATE_ATTACH.  These are initial states for these class instances and I am able to drive their states from the beginning using the ready() methods for both classes.

The issue I have noticed is that there is no way to reset the GSMClient instance state to its initial state CLIENT_STATE_IDLE following a reset of the GSM and GPRS class instances noted above.

My script uses an asynchronous GSMClient instance GSMClient::GSMClient(false) with my own time loop to establish a connection to the server.  If the GSMClient::connect(server, port) hangs on the AT+USOCR=6 command the GSMClient _state is stuck in the CLIENT_STATE_WAIT_CREATE_SOCKET_RESPONSE state.  Therefore, when I drive the GSMClient state machine with a GSMClient::connect() the GSMClient state machine reads a residual socket from the _response data and transitions to the CLIENT_STATE_CONNECT state.  When my script issues the next GSMCLient::ready() an AT+USOCO=... command is issued.  This ends in error because the socket is obviously no longer valid.  I am able to work around this situation by issuing GSMClient::ready() calls until the invalid socket is closed and the GSMClient state returns to CLIENT_STATE_IDLE.

Looking at the GSMClient code it looks like my issue would be addressed by changing the logic at the beginning of the GSMClient::connect() method.  Currently it looks like this:

 if (_synch) {
    while (ready() == 0);
  } else if (ready() == 0) {
    return 0;
  }

 _state = CLIENT_STATE_CREATE_SOCKET;

It appears the purpose of this code is to ensure that no other AT command is outstanding to the modem.  Unfortunately, if there are no commands outstanding to modem the ready() calls drive the GSMClient state machine.  If the ready() calls were replaced with MODEM.ready() calls that would ensure that there are no outstanding commands to the modem and wouldn't drive the GSMClient state machine before it is reset to CLIENT_STATE_CREATE_SOCKET.

Nels52, 2018-10-04:
Hi @sandeepmistry,

Does the change in the GSMClient::connect() method that I recommended near the end of my long winded explanation look like a good one?  I have tested it and found that it brought my GSMClient back to the initial CLIENT_STATE_CREATE_SOCKET state while still ensuring that there were no outstanding AT commands to the modem.

sandeepmistry, 2018-10-04:
@Nels52 I think so, however would you be able to provide a minimal example sketch to reproduce your issue. This would really clear things up. Please create a new issue for this topic, as from my understanding so far it's not entirely related to the original issue topic. Thanks!

Nels52, 2018-10-04:
Hi @sandeepmistry,

Not sure how much of my script to include so I have included all of it:

`/*
  HomeMonitor V2

 This sketch reads a temperature monitor every N minutes and saves the last M temperatures read. 
 A message to the Hologram web server is generated if one of the following conditions has been met:
         (1) M intervals have passed since the last message was sent to Hologram.  Report lowest temperature read in last M read intervals.
         (2) The temperature read for this interval has dropped below 0 F.
         (3) The temperature read for this interval has risen to 100 F or above.
         (4) The temperature read for this interval has fallen below 50 F for the first time in the last M intervals.
         (5) The temperature read for this interval has fallen below the last temperature reported to the server AND the temperature is below 50 F.
 
 Circuit:
 * MKR GSM 1400 board
 * Antenna
 * SIM card with a data plan
 * Temperature sensor

 created 9-24-2018
 by Bob Nelson
*/

// libraries
#include <MKRGSM.h>

// Replace "arduino_secrets.h" as all the information for that is publicly available from https://hologram.io/docs/guide/connect/connect-device/#apn-settings

// Hologram SIM does not use a PIN number
const char PINNUMBER[] = " ";

// APN data. Hologram Web Server does not use a login or password.
const char GPRS_APN[] = "hologram";
const char GPRS_LOGIN[] = " ";
const char GPRS_PASSWORD[] = " ";
String HOLOGRAM_API_KEY = ",rwm6,xM";

// Special Messages and TOPICs sent to Hologram server using the JSON string format with Hologram's embedded API.
String TOPIC_LOW_TEMP = "LOW_TEMP";
String TOPIC_HIGH_TEMP = "HIGH_TEMP";
String TOPIC_WARNING_TEMP = "WARNING_TEMP";
String TOPIC_TEMP = "TEMPERATURE";
String TOPIC_RESTART = "RESTART";
String MESSAGE_LOW_TEMP = "Temperature below 0 F";
String MESSAGE_HIGH_TEMP = "Temperature above 99 F";

// Create an array of Strings to report Temperature values.  This eliminates the need to allocate/free system memory
// when reporting temperature.
String TEMP_ARRAY[] = { "0 F",  "1 F",  "2 F",  "3 F",  "4 F",  "5 F",  "6 F",  "7 F",  "8 F",  "9 F",
                       "10 F", "11 F", "12 F", "13 F", "14 F", "15 F", "16 F", "17 F", "18 F", "19 F",
                       "20 F", "21 F", "22 F", "23 F", "24 F", "25 F", "26 F", "27 F", "28 F", "29 F",
                       "30 F", "31 F", "32 F", "33 F", "34 F", "35 F", "36 F", "37 F", "38 F", "39 F",
                       "40 F", "41 F", "42 F", "43 F", "44 F", "45 F", "46 F", "47 F", "48 F", "49 F",
                       "50 F", "51 F", "52 F", "53 F", "54 F", "55 F", "56 F", "57 F", "58 F", "59 F",
                       "60 F", "61 F", "62 F", "63 F", "64 F", "65 F", "66 F", "67 F", "68 F", "69 F",
                       "70 F", "71 F", "72 F", "73 F", "74 F", "75 F", "76 F", "77 F", "78 F", "79 F",
                       "80 F", "81 F", "82 F", "83 F", "84 F", "85 F", "86 F", "87 F", "88 F", "89 F",
                       "90 F", "91 F", "92 F", "93 F", "94 F", "95 F", "96 F", "97 F", "98 F", "99 F"};

String HOLOGRAM_RESPONSE_ARRAY[] = {"Send successful, no error",
                                    "Connection was closed so we couldn’t read enough",
                                    "Couldn’t parse the message",
                                    "Auth section of message was invalid",
                                    "Payload type was invalid",
                                    "Protocol type was invalid",
                                    "An internal error occurred",
                                    "Metadata section of message was formatted incorrectly",
                                    "Topic contained invalid characters or was too long"};
[... 492 lines cut ...]
          Serial.print("Unrecognized Response: ");
          Serial.println(c);
        }
        
        bNextCharIsResponseCode = false;
      } else if (c == '[')
      {
        bNextCharIsResponseCode = true;
      }
    }
  }
}
`

The problem occurs as follows:
  1. connectClient() in script fails when AT+USOCR=6 to modem hangs.
  2. startWebClient() in script is called to recover by invoking GSM:begin() and GPRS::attachGPRS().
  3. connectClient() is called again, however, GSMClient has been left in 
      CLIENT_STATE_WAIT_CREATE_SOCKET_RESPONSE state causing GSMClient::connect() to issue a 
      AT+USOCO=... command without a valid socket.
  4. MOdem returns ERROR to  AT+USOCO=... command.
  5. The next GSMClient::ready() call generates a AT+USOCL command which also ends in ERROR 
      because there is no valid socket.
  6. At this point another GSMClient::connect() can be successfully issued because the GSMClient state 
      has been  returned to CLIENT_STATE_IDLE.

Nels52, 2018-10-04:
Hi @sandeepmistry,

Not sure which part of my sketch to include so I just attached the whole thing.
[HomeMonitorV2.txt](https://github.com/arduino-libraries/MKRGSM/files/2447953/HomeMonitorV2.txt)

I will open another issue but here is the summary:

 1. All of my modem interface instances (GSM, GPRS, and GSMClient) are configured to operate 
     asynchronously so my script can control the progression thru their state machines using the ready() 
     methods in each class.  As noted by this issue this prevents unrecoverable hangs when no response is 
     received from the modem to an AT command.
 2. The connectClient() function in my script fails because the AT+USOCR=6 command issued by 
     GSM::connect() method hangs attempting to open a socket.
 3. My script attempts to recover by calling the startWebClient() which invokes GSM::begin() followed by 
     GPRS::attachGPRS().
 4. If both GSM and GPRS are restarted successfully connectClient() is called again to issue a 
     GSMClient::connect().
 5. The GSMClient::connect() issues an AT+USOCO=... command because the GSMClient has been left in 
     the CLIENT_STATE_WAIT_CREATE_SOCKET_RESPONSE state by the modem hang in step 2.
 6. The AT+USOCO=... command returns ERROR because the socket is not valid.
 7. The next GSMClient::ready() method call generates an AT+USOCL command which also returns 
     ERROR, however, the GSMClient state has been returned to CLIENT_STATE_IDLE so that the next 
     GSMClient::connect() method call will work properly.

Hope this helps.  Let me know if you need something else.  In the mean time I will recreate the issue with the original MKRGSM 1.2.1 library and post the serial output.

Melonfruit, 2019-01-17:
@lathoub Thanks for replying. Yes, I would be very interested in your sketch. If you wouldn't mind, could you also tell me how you powered your MKR1400 while you were testing it through Arduino IDE? Thank you again

Nels52, 2019-01-17:
@Melonfruit 

I am using a HomeSpot 5V power supply/adapter/charger which is capable of delivering up to 2.4 Amps.  It has a USB interface which I cut off and exposed the ground and hot wires so I could insert them in my breadboard with the MKRGSM 1400.  I found it on Amazon.

Here is my sketch.

`/*
  HomeMonitor V5

 This sketch reads a temperature monitor every N minutes and saves the last M temperatures read. 
 A message to the Hologram web server is generated if one of the following conditions has been met:
         (1) M intervals have passed since the last message was sent to Hologram.  Report lowest temperature read in last M read intervals.
         (2) The temperature read for this interval has dropped below 0 F.
         (3) The temperature read for this interval has risen to 100 F or above.
         (4) The temperature read for this interval has fallen below 50 F for the first time in the last M intervals.
         (5) The temperature read for this interval has fallen below the last temperature reported to the server AND the temperature is below 50 F.

Changes from HomeMonitorV3:
         - Must use MKRGSM 1400 1.3.0 library.
         - Use the new GSMClient socket buffer feature which reads up to 512 bytes of data from the modem in the AT+USORD= command
         - Modify GSMClient::stop() in the MKRGSM 1400 1.3.0 library to reset the GSMClient state at the top of the stop() method:    
               
                _state = CLIENT_STATE_IDLE;                

Changes from HomeMonitorV4:
         - Use MKRGSM 1400 1.3.1 library which contains my GSMClient::stop() change

 Circuit:
 * MKR GSM 1400 board
 * Antenna
 * SIM card with a data plan
 * Temperature sensor

 created 10-26-2018
 by Bob Nelson

*/

// libraries
#include <MKRGSM.h>

// Replace "arduino_secrets.h" as all the information for that is publicly available from https://hologram.io/docs/guide/connect/connect-device/#apn-settings

// Hologram SIM does not use a PIN number
const char PINNUMBER[] = " ";

// APN data. Hologram Web Server does not use a login or password.
const char GPRS_APN[] = "hologram";
const char GPRS_LOGIN[] = " ";
const char GPRS_PASSWORD[] = " ";
String HOLOGRAM_API_KEY = "xxxxx";

// Special Messages and TOPICs sent to Hologram server using the JSON string format with Hologram's embedded API.
String TOPIC_LOW_TEMP = "LOW_TEMP";
String TOPIC_HIGH_TEMP = "HIGH_TEMP";
String TOPIC_WARNING_TEMP = "WARNING_TEMP";
String TOPIC_TEMP = "TEMPERATURE";
String TOPIC_RESTART = "RESTART";
String MESSAGE_LOW_TEMP = "Temperature below 0 F";
String MESSAGE_HIGH_TEMP = "Temperature above 99 F";

// Create an array of Strings to report Temperature values.  This eliminates the need to allocate/free system memory
// when reporting temperature.
String TEMP_ARRAY[] = { "0 F",  "1 F",  "2 F",  "3 F",  "4 F",  "5 F",  "6 F",  "7 F",  "8 F",  "9 F",
                       "10 F", "11 F", "12 F", "13 F", "14 F", "15 F", "16 F", "17 F", "18 F", "19 F",
                       "20 F", "21 F", "22 F", "23 F", "24 F", "25 F", "26 F", "27 F", "28 F", "29 F",
                       "30 F", "31 F", "32 F", "33 F", "34 F", "35 F", "36 F", "37 F", "38 F", "39 F",
                       "40 F", "41 F", "42 F", "43 F", "44 F", "45 F", "46 F", "47 F", "48 F", "49 F",
                       "50 F", "51 F", "52 F", "53 F", "54 F", "55 F", "56 F", "57 F", "58 F", "59 F",
                       "60 F", "61 F", "62 F", "63 F", "64 F", "65 F", "66 F", "67 F", "68 F", "69 F",
[... 497 lines cut ...]
        {
          Serial.print("Unrecognized Response: ");
          Serial.println(c);
        }
        
        bNextCharIsResponseCode = false;
      } else if (c == '[')
      {
        bNextCharIsResponseCode = true;
      }
      
      availableBytesToRead--;
    }
    
    loopCount++;
    delay(50);        // Delay 50 ms before the next AT+USORD=0,512 command issued by the GSMClient connected() method
  } while (loopCount < 600);

  // Did not receive response and/or close from Hologram server.  Return false so modem and client can be restarted.
  Serial.println();
  Serial.println("No response/close from Hologram.");
  client.stop();
  clientConnected = false;
  return false;
}`

Nels52, 2019-01-17:
@Melonfruit 

My sketch has some strange highlighing when I inserted it.  The key routines are startWebClient(), connectClient(), and checkForResponseFromHologram() which drive the state machines asynchronously.

Melonfruit, 2019-01-18:
@Nels52 Thanks you for your help. I bought an 3.7 LiPo 2000mAh battery that I connected to the VIN and GND pins on the MKR1400 while its connected to my PC. I've also tried using the 5V ping with the GND. Am I doing anything wrong with regards to this setup?

Nels52, 2019-01-18:
@Melonfruit 
I connect my 5V external power supply to VIN as documented for the MKR GSM 1400.  I have not used a LiPo 3.7V battery but it was my impression that it used a separate connection.  The VIN pin appears to be designed for a 5V power supply and the 5V pin is strictly for output.