arduino-libraries/MKRGSM #31: receiving SMS with characters äöüÄÖÜ,...

johannstieger, 2018-07-10:
I can not receive special characters in SMS used in german (äöüÄÖÜß,... - I wanna receive SMS and forward it with GSMSSLClient to a web-server). When I send theses characters to my MKRGSM1400 I get nothing/the characters are missing.
Is it possible to extend the library to receive either all 7-bit characters included in the [GSM 7-bit default Alphabet]([https://en.wikipedia.org/wiki/GSM_03.38]) (so I can translate the 7-bit GSM characters to my own 8-bit or UTF charset) or to extend the library so I can select the charset as a parameter in the library and the MKRGSM library does the rest?

Rocketct, 2018-07-16:
Hi @johannstieger we doesn't support the extended  character set because the read() function use a buffer of byte, for your issue, however, you could implement  a logic that handles the unicod code points from the _incomingBuffer in the GSM SMS class, and recognize the characters that you need.

sandeepmistry, 2018-09-12:
Closing this for now, as there is not been any other feedback for this feature.

closed 2018-09-12 by sandeepmistry