arduino-libraries/MKRGSM #35: MKR 1400 Modem Hangs After Repeated Use

alanoatwork, 2018-07-26:
My sketch is pretty straightforward... I look for an incoming SMS, and if there is one, I send it back out on the Hologram.io network.  It works fine until a few hours goes by and then the modem returns ERROR.  Here's the sketch.

```
#include <MKRGSM.h>
const char PINNUMBER[] = " ";
const char GPRS_APN[] = "hologram";
const char GPRS_LOGIN[] = " ";
const char GPRS_PASSWORD[] = " ";

String HOLOGRAM_DEVICE_KEY = "******";
String HOLOGRAM_TOPIC = "_SOCKETAPI_";

GSMClient client;
GPRS gprs;
GSM gsm(1);                                             // Enable debug
GSM_SMS sms;

char server[] = "cloudsocket.hologram.io";
int port = 9999;
boolean isSMSAvailable = false;
char sms_message[145];

void setup() {
  Serial.begin(115200);
  while(!Serial);

  connectGSM();   
}

void connectGSM() {
  boolean connected = false;

  while (!connected) {
    Serial.println("Begin GSM Access");
      
    if ((gsm.begin() == GSM_READY) &&
        (gprs.attachGPRS(GPRS_APN, GPRS_LOGIN, GPRS_PASSWORD) == GPRS_READY)) {
      connected = true;
      Serial.println("GSM Access Success");
    } 
    else {
      Serial.println("Not connected");
      delay(1000);
    }
  }
}

void loop() {
  // Get any new incoming txt messages
  int c;
  if (sms.available()) {
    int i = 0;
    while ((c = sms.read()) != -1) {
      sms_message[i++] = (char)c;
    }
    sms_message[i] = '\0';        // Terminate message
    isSMSAvailable = true;
    sms.flush();
  }

  // Send message back through hologram
  if(isSMSAvailable) {
    isSMSAvailable = false;

    if (client.connect(server, port)) {
      client.print("{\"k\":\"" + HOLOGRAM_DEVICE_KEY + "\",\"d\":\"");
      client.print(sms_message);
      client.println("\",\"t\":\""+HOLOGRAM_TOPIC+"\"}");
      client.stop();
    }
  }

  delay(5000);
}
```
When it works fine, the output looks like this:

```
AT+CMGL="REC UNREAD"

+CMGL: 19,"REC UNREAD","+19495472010",,"18/07/26,00:57:17+04"
Tgghgfh

OK
AT+CMGD=19

OK
AT+USOCR=6

+USOCR: 0

OK
AT+USOCO=0,"cloudsocket.hologram.io",9999

OK
AT+USOWR=0,21,"7B226B223A22433E383375242B57222C2264223A22"

+USOWR: 0,21

OK
AT+USOWR=0,7,"54676768676668"

+USOWR: 0,7

OK
AT+USOWR=0,20,"222C2274223A225F534F434B45544150495F227D"

+USOWR: 0,20

OK
AT+USOWR=0,2,"0D0A"

+USOWR: 0,2

OK
AT+USOCL=0

OK
AT+CMGL="REC UNREAD"

OK
```
When it fails, here's what happens:

```
AT+CMGL="REC UNREAD"

OK
AT+CMGL="REC UNREAD"

OK
AT+CMGL="REC UNREAD"

+CMGL: 19,"REC UNREAD","+19495472010",,"18/07/26,03:09:53+04"
Testing one two three

OK
AT+CMGD=19

OK
AT+USOCR=6

ERROR
AT+CMGL="REC UNREAD"

OK
```

Any ideas?

FrancMunoz, 2018-07-26:
Hi @alanoatwork , I looked at your sketch and I see a some bad practices that led me to the same situation in my own ones. 

SARA returns ERROR when you try open a socket by some causes... the most common are modem is not attached or there's no signal and so on... So, to avoid this, I suggest you before try to connect the socket, just check if all is ok... GSM and GPRS, then open the socket and if it stills fail just retry leaving modem its time to update all its internal registries.

The connection could be lost for some causes... DHCP timeout, provider strategi, etc. In my case Movistar (from Spain) is closing connections every 24 hours. So a good strategy reconnecting is recommended.

If you take a look at #30 , I posted an example that worked for me. It's about a persistent connection but I think it would help you about checking and reconnecting. 

We have to remember that re-connection it's not done automatically so our logic has to contemplate that.

All in all, in your sketch if connection to server fails the message is lost. I'd suggest to delete SMS once is sent to server or retain locally until is sent. I tell you that because in my experience internet connection fluctuates a lot.

Well, it's all based in my own experience... hope it helps!

alanoatwork, 2018-07-30:
Franc,

In the main loop I added the following code:

```
 if(gsm.isAccessAlive()) {
    if(gprs.status() != GPRS_READY) {
      if(gprs.attachGPRS(GPRS_APN, GPRS_LOGIN, GPRS_PASSWORD) == GPRS_READY)
        Serial.println("GPRS ready!");
      else
        Serial.println("GRPS not ready!");
    }
  }
  else {
    Serial.println("Reconnect to GSM...");
    connectGSM();
  }
```
So far my board seems to be working for at least 24 hours now. I'm reluctant to open a socket and send messages on a regular basis as this will consume data. I am curious to know if there are any other items I should be checking to make sure the next socket connection won't result in an ERROR response.

FrancMunoz, 2018-07-31:
Hi @alanoatwork ,

I remember when I got the same problem as you, I changed to persistent connection to server to avoid open and closing socket. SARA can open up to 7 sockets, and I think the problem could be that sockets aren't completely closed so SARA doesn't have enough stack to open a new socket and returns ERROR.

The socket number is returned when creating a socket:

```
OK
AT+USOCR=6

+USOCR: <SOCKET NUMBER>

OK
```

Could you enable debug and just before it fails check what socket number is the last that worked? In my case sometimes sockets were not closed and SARA can't connect.

Also in your code add the following line in `client.connect` if:

```c++
 // Send message back through hologram
  if(isSMSAvailable) {
    isSMSAvailable = false;

    if (client.connect(server, port)) {
      client.print("{\"k\":\"" + HOLOGRAM_DEVICE_KEY + "\",\"d\":\"");
      client.print(sms_message);
      client.println("\",\"t\":\""+HOLOGRAM_TOPIC+"\"}");
      client.stop();
    } else {
       //GET LAST SOCKET ERROR DEBUG MUST BE ENABLED TO SEE RESULT
      MODEM.send("AT+USOER");
    }
  }
```

Could you please test and share the results? Or check in manual page 767.

[AT Command Manual](https://www.u-blox.com/sites/default/files/u-blox-CEL_ATCommands_%28UBX-13002752%29.pdf)

Also I suggest you to remove `delay(5000)` and work with `millis()` letting the loop be faster. Also I'd suggest to check and read server response, I think this would cause that socket doesn't close properly. But it's only intuition...

I noticed, that if MKR doesn't have enough power it fails during socket creation and GPRS attach. I guess you are aware of this but as it's very important I recall.

Hope it helps!

alanoatwork, 2018-08-02:
Franc,

I removed the delay(5000) and added the additional debug code.  Your point regarding power is a good one. I don't have a LiPo battery connected, but now I have one on order and will resume my testing once I have it.

Will report soon...

alanoatwork, 2018-09-13:
Still testing, with and without the modem debug enabled. Sorry I got sidetracked but I'll report back soon.

alanoatwork, 2018-10-14:
Since I  used a 1500 mAHr battery I haven't been able to duplicate my issue, so let's close this out. Thanks for your patience.

closed 2018-10-15 by sandeepmistry