arduino-libraries/MKRGSM #40: Allow SSL connection without certs

Josca, 2018-09-01:
Hi,

 I need to connect to my HTTPS website without valid certificate from my MKR GSM 1400. I use GSMSSLClient for that. I see there is no option for this currently. I would like to add something like `-k` switch in `curl` or `verify` param in python `requests`.

Could you recommend me how to do that?

jgartner1, 2018-09-05:
If your certificate is not valid why even connecting over an unsafe connection like that, wich can get interferred pretty much as easily as using regular http connections? Just connect via port 80 instead of 443 and ensure the server where the request goes to supports port 80 aswell and doesn't just redirect you to 443 (https). From a security point of view it's equally safe. 

Or if you want true encryption I suggest you create yourself a free valid certificate signed by a CA. For this I would recommend LetsEncrypt. With an online-tool like https://www.sslforfree.com/ you can even create your signed certificates online without instally any software on your server.

Josca, 2018-09-05:
HTTPS connection is secure (encrypted) even without valid certificate see, so it's better then HTTP.

I already tried LetsEncrypt using command line tool, but I was not successful. It ended with errors. Maybe it's not possible get valid certificate for long domain like `ec2-35-138-22-53.eu-central-1.compute.amazonaws.com`.

FrancMunoz, 2018-09-05:
Hi!

SARA modem is configured by default to not validate certificates, the AT command used for configure that is +USECPRF, see [manual](https://www.u-blox.com/sites/default/files/u-blox-CEL_ATCommands_%28UBX-13002752%29.pdf) at page 629.

Sending AT+USECPRF=0,0,0 to modem will change certificate validation level to no validation. All in all this is the default option, so nothing will be changed:

> 0: certificate validation level; allowed values for <param_val>:
> - 0 (factory-programmed value): **Level 0 - No validation; the server certificate will not be checked or**
> **verified. The server in this case is not authenticated.**
> - 1: Level 1 - Root certificate validation without URL integrity check. The server certificate will be verified
> with a specific trusted certificates or with each of the imported trusted root certificates.
> - 2: Level 2 - Root certificate validation with URL integrity check. Level 1 validation with an additional
> URL integrity check.
> - 3: Level 3 - Root certificate validation with check of certificate validity date. Level 2 validation with an
> additional check of certificate validity date.
> 

I guess your problem is about the function called to connect. If you want to use SSL you should use `GSMClient::connectSSL` instead of `GSMClient::connect`.

I've just make a PR #42 that is basically a function to download a string using HTTP or HTTPs, just take a look at the PR. 

I've used SSL using `GSMClient::connectSSL` and it worked for me without issues.

Rocketct, 2018-10-18:
Hi all, @Josca, we doesn't support SSL connection without certs, with the PR https://github.com/arduino-libraries/MKRGSM/pull/56  we set this  level of certificates validation: 

> 1: Level 1 - Root certificate validation without URL integrity check. The server certificate will be verified

you can find the certificates in https://github.com/arduino-libraries/MKRGSM/blob/master/src/utility/GSMRootCerts.h

i'll close the issue.

closed 2018-10-18 by Rocketct

darleys, 2020-01-02:
Still, this is not resolving https sites, What is the process to access https . The GSMRootCerts.h is not sufficient for many websites.

beenjamminnallday, 2021-02-15:
Hello all, I am using an MKRGSM 1400 and attempting to connect on a website a friend on mind created. So far the website is supposed to only return {"message":"Success","status":200}. 
I had issues even connecting to the server but I was able to resolve this by using _MODEM.sendf("AT+USECPRF=0,0,0");_ to change the modem certificate validation level to no validation. 
However, another issue comes up where when attempting to connect to a https server/website and perform a GET request, I get _301 Moved Permanently_. My hypothesis was that it is performing the HTTP request but cannot/does not redirect me to the website. 

Unfortunately, I am no longer able to get this error message. Any help on this issue would be appreciated. I am simply trying to connect to a web server, i'm a relative beginner but I assume this should be straight forward and is supported by the device.

Attached is the code and my serial output.

```
/*
  Web client

 This sketch connects to a website using SSL through a MKR GSM 1400 board. Specifically,
 this example downloads the URL "http://www.arduino.cc/asciilogo.txt" and
 prints it to the Serial monitor.

 Circuit:
 * MKR GSM 1400 board
 * Antenna
 * SIM card with a data plan

 created 8 Mar 2012
 by Tom Igoe
*/

// libraries
#include <MKRGSM.h>

#include "arduino_secrets.h" 
// Please enter your sensitive data in the Secret tab or arduino_secrets.h
// PIN Number
const char PINNUMBER[]     = SECRET_PINNUMBER;
// APN data
const char GPRS_APN[]      = SECRET_GPRS_APN;
const char GPRS_LOGIN[]    = SECRET_GPRS_LOGIN;
const char GPRS_PASSWORD[] = SECRET_GPRS_PASSWORD;

// initialize the library instance
GSMSSLClient client;
GPRS gprs;
GSM gsmAccess(true);

//URL, path and port (for example: arduino.cc)
char server[] = "dashboard.instant-alert.ca";
char path[] = "/";
int port = 443; // port 443 is the default for HTTPS

void setup() {
  // initialize serial communications and wait for port to open:
  Serial.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }
 MODEM.debug();
  Serial.println("Starting Arduino web client.");
  // connection state
  bool connected = false;

  // After starting the modem with GSM.begin()
  // attach the shield to the GPRS network with the APN, login and password
  while (!connected) {
    if ((gsmAccess.begin(PINNUMBER) == GSM_READY) &&
        (gprs.attachGPRS(GPRS_APN, GPRS_LOGIN, GPRS_PASSWORD) == GPRS_READY)) {
      connected = true;
    } else {
      Serial.println("Not connected");
      delay(1000);
    }
  }

[... 361 lines cut ...]

OK
AT+USOCR=6

+USOCR: 0

OK
AT+USOSEC=0,1,0

OK
AT+USECPRF=0,0,1

OK
AT+USOCO=0,"dashboard.instant-alert.ca",443

ERROR

+UUSOCL: 0
AT+USOCL=0

ERROR
connection failed

disconnecting.
```