arduino-libraries/MKRGSM #45: Compile warning

jklein131, 2018-09-17:
MKR GSM compiles with a warning on the ARDUINO DESKTOP IDE 
"C:\Users\joshuak\AppData\Local\Temp\arduino_build_385205\libraries\MKRGSM\Modem.cpp.o"
C:\Users\joshuak\Documents\Arduino\libraries\MKRGSM\src\Modem.cpp: In member function 'int ModemClass::autosense(int)':

C:\Users\joshuak\Documents\Arduino\libraries\MKRGSM\src\Modem.cpp:111:61: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]

   for (unsigned long start = millis(); (millis() - start) < timeout;) {

                                                                                     ^

sandeepmistry, 2018-09-19:
Hi @jklein131,

Thanks for pointing this out! There's a change in pull request #44 to fix this: https://github.com/arduino-libraries/MKRGSM/pull/44/commits/6e1ced41f9d5a9ba3f04869c50bd050e5c6decec

sandeepmistry, 2018-09-27:
Closed via #44.

closed 2018-09-27 by sandeepmistry