arduino-libraries/MKRGSM #55: GSMClient::connect() drives GSMClient state machine before initializing _state to CLIENT_STATE_CREATE_SOCKET;

Nels52, 2018-10-04:
If the modem fails to respond to an AT command the GSMClient state machine hangs in a state other than CLIENT_STATE_CREATE_SOCKET.  After the modem is reinitialized with GSM::begin() and GPRS::attachGPRS() methods the GSMClient::connect() method returns ERROR because this method fails to reset the state to CLIENT_STATE_CREATE_SOCKET before driving the GSMClient state machine.

The problem is detailed below:

1. All of my modem interface instances (GSM, GPRS, and GSMClient) are configured to operate 
     asynchronously so my script can control the progression thru their state machines using the ready() 
     methods in each class.  This prevents unrecoverable hangs when no response is received from the 
     modem to an AT command.
 2. Script issues GSMClient::connect() which generates an AT+USOCR=6 command to open a socket.
 3. Script attempts to recover by invoking GSM::begin() followed by GPRS::attachGPRS().
 4. If both GSM and GPRS are restarted successfully GSMClient::connect() is invoked again.
 5. The GSMClient::connect() issues an AT+USOCO=... command because the GSMClient has been left in 
     the CLIENT_STATE_WAIT_CREATE_SOCKET_RESPONSE state by the modem hang in step 2.
 6. The AT+USOCO=... command returns ERROR because the socket is not valid.
 7. The next GSMClient::ready() method call generates an AT+USOCL command which also returns 
     ERROR, however, the GSMClient state has been returned to CLIENT_STATE_IDLE so that the next 
     GSMClient::connect() method call will work properly.

I have changed the GSMClient::connect() method to address this issue and it works without any adverse affects.  The change is a simple one to the beginning of the GSMClient::connect() method:

Here is the beginning of the GSMClient::connect() version in the 1.2.1 library:

 if (_synch) {
    while (ready() == 0);
  } else if (ready() == 0) {
    return 0;
  }
  _state = CLIENT_STATE_CREATE_SOCKET;

I have made the following changes:

 if (_synch) {
    while (MODEM.ready() == 0);
  } else if (MODEM.ready() == 0) {
    return 0;
  }
  _state = CLIENT_STATE_CREATE_SOCKET;

By replacing the ready() calls with MODEM.ready() calls the GSMClient state machine is not erroneously driven before the _state is initialized.  Therefore, the method still returns 0 in asynchronous mode OR stays in a while loop in synchronous mode if an AT command is outstanding.  Normally this will not be the case but I understand the need to check for an outstanding AT command before continuing with the connect request.

NOTE: I have not tested synchronous mode but looking at the code change I don't see an issue.

Nels52, 2018-10-05:
Hi @sandeepmistry,

I have attached my script (HomeMonitorV2,txt) and the Serial output (SerialOutput.txt)
from that script with debug enabled.  In the serial output you will see where my script times out a connection request: "Connection Timed OUT loopCount = 600".  After the GSM and GPRS have been restarted you will see where a connnection is attempted: "GSMClient state machine out of sync".  Because the GSMClient is in the CLIENT_STATE_WAIT_CREATE_SOCKET_RESPONSE state the AT+USOCO=0,"cloudsocket.hologram.io",9999 command is issued with an invalid socket.  The modem returns ERROR to this command and the AT+USOCL=0 command that follows.  When the GSMClient::connect() is issued after these ERROR returns a connection is successfully established because the GSMClient has been set the initial CLIENT_STATE_CREATE_SOCKET state.

[HomeMonitorV2.txt](https://github.com/arduino-libraries/MKRGSM/files/2451396/HomeMonitorV2.txt)
[SerialOutput.txt](https://github.com/arduino-libraries/MKRGSM/files/2451397/SerialOutput.txt)

Rocketct, 2018-10-22:
Hi @Nels52, thank you for open the issue we are work on this, i should ask a more minimal sketch that reproduce the error?

Nels52, 2018-10-22:
Hi @Rocketct, I have embedded a stripped down version of my script.  I have compiled it but not loaded and tested it but it should recreate the issue.  The key to recreation is to connect to a web server, send a message and disconnect from the web server.  In my case I use Hologram which automatically disconnects after receiving a message.  It most likely will take multiple iterations before the client connect sequence hangs.  A little more than a week ago my connect sequence would hang several times a day.  However, it appears something has changed with my network (changes/fixes by my network provider??) and now my client connect sequence hasn't failed for a week.

Hope this helps.  Let me know if you need further assistance.

/*
 Reproduce GSMClient Connect Issue

 This script sends a test message to a web server (in my case Hologram) every 10 minutes.
 After sending the test message the Hologram web server disconnects.  Therefore the connect sequence must
 be reinitiated every time the test message is sent.  If you use a different web server you must ensure that
 the connection is terminated each time a message is sent.

 The issue occurs when the connection sequence hangs (usually on the AT+USOCR=6 command). When this happens the
 script restarts GSM and GPRS from scratch (GSM.begin() and GPRS.attachGPRS()).  After GSM and GPRS are restarted
 successfully the GSMClient.connect(server, port) fails (ERROR is returned to next AT command sent to the modem)
 because the GSMClient state has not been reinitialized.

 In summary:
  - Use GSM, GPRS, and GSMClient instances running in asynchronous mode.
  - Restart GSM and GPRS instances after a GSMClient connect sequence hangs.
  - Reconnect GSMClient after successfully restarting GSM and GPRS.
 
 Circuit:
 * MKR GSM 1400 board
 * Antenna
 * SIM card with a data plan

*/

// libraries
#include <MKRGSM.h>

// Replace "arduino_secrets.h" as all the information for that is publicly available from https://hologram.io/docs/guide/connect/connect-device/#apn-settings

// Hologram SIM does not use a PIN number
const char PINNUMBER[] = " ";

// APN data. In my example I use the hologram web server.  However, any web server can be used.
const char GPRS_APN[] = "hologram";
const char GPRS_LOGIN[] = " ";
const char GPRS_PASSWORD[] = " ";
String HOLOGRAM_API_KEY = "xxxxxx";   // Need your own hologram generated key here

String TOPIC = "MY_TOPIC";
String MESSAGE_TEST = "Test Message";

String HOLOGRAM_RESPONSE_ARRAY[] = {"Send successful, no error",
                                    "Connection was closed so we couldn’t read enough",
                                    "Couldn’t parse the message",
                                    "Auth section of message was invalid",
                                    "Payload type was invalid",
                                    "Protocol type was invalid",
                                    "An internal error occurred",
                                    "Metadata section of message was formatted incorrectly",
                                    "Topic contained invalid characters or was too long"};

/// Test multiple connections within one run
// initialize the library instances
GSMClient client(false);     // Instantiate a GSMClient instance operating in asynchronous mode
GPRS gprs;
GSM gsmAccess(true);         // Turn debug on so we see all AT commands and responses

// Hologram's Embedded API (https://hologram.io/docs/reference/cloud/embedded/) URL and port
char server[] = "cloudsocket.hologram.io";
int port = 9999;

bool clientConnected = false;     // Connection state flag
bool restartClient = true;        // Client restarted flag - initially set to ensure temperature is reported at startup
bool sendTestMessage = false;      // Send test message
unsigned long lastConnectTime = 0;    // Last time temperature was read
[... 293 lines cut ...]
          Serial.println(HOLOGRAM_RESPONSE_ARRAY[c - '0']);
        } else 
        {
          Serial.print("Unrecognized Response: ");
          Serial.println(c);
        }
        
        bNextCharIsResponseCode = false;
      } else if (c == '[')
      {
        bNextCharIsResponseCode = true;
      }
    }
    
    loopCount++;
    delay(100);
  } while (loopCount < 600);

  // Did not receive response and/or close from Hologram server.  Return false so modem and client can be restarted.
  Serial.println();
  Serial.println("No response/close from Hologram.");
  client.stop();
  clientConnected = false;
  return false;
}

Rocketct, 2018-10-24:
Hi @Nels52 i had test the sketch, actually i never get the error and the GSM work properly, how is going your test? in this day was released a new version of the Library (1.3.0) could you test with this one and let us know the results?

Nels52, 2018-10-24:
Hi @Rocketct.  I have figured out a way to recreate the issue:

1. Use the script above with GSM, GPRS, and GSMClient instances operating in asynchronous mode.
2. Run the script and let it sent one message successfully. Make sure that the server you are using disconnects after it sends a response just like Hologram does.
3. Before the next message is sent to the server disconnect the antenna.
4. Watch the connection sequence fail as shown in the included trace.
5. Reconnect the antenna and watch GSM and GPRS initialize successfully.
6. The GSMClient connect sequence will return errors because the GSMClient state was left in a residual state from the previous connection attempt.

I have included an annotated trace with my recreation of the issue.

*************************************************************************************************************
*  Here is an example of a good send sequence to the Hologram server.  Note that after the
*  Hologram server sends a response it disconnects.
*************************************************************************************************************

13:51:55.370 -> connecting...
AT+USOCR=6

13:51:55.474 -> +USOCR: 0
13:51:55.474 -> 
13:51:55.474 -> OK
AT+USOCO=0,"cloudsocket.hologram.io",9999

13:51:58.903 -> OK
13:51:58.903 -> Connect after 3500 ms
13:51:58.903 -> connected
13:51:58.903 -> Report Temperature: 68 F
13:51:58.903 -> AT+USOWR=0,45,"7B226B223A22536339425E5D2E70222C2264223A2236382046222C2274223A2254454D5045524154555245227D"

13:51:58.903 -> +USOWR: 0,45
13:51:58.903 -> 
13:51:58.903 -> OK
13:51:58.903 -> AT+USOWR=0,2,"0D0A"

13:51:58.903 -> +USOWR: 0,2
13:51:58.903 -> 
13:51:58.903 -> OK
13:52:00.015 -> +UUSORD: 0,5
13:52:00.015 -> AT+USORD=0,1

13:52:00.015 -> +USORD: 0,1,"5B"
13:52:00.015 -> 
13:52:00.015 -> OK
13:52:00.015 -> [
13:52:00.116 -> +UUSORD: 0,4
13:52:00.116 -> AT+USORD=0,1

13:52:00.116 -> +USORD: 0,1,"30"
13:52:00.116 -> 
13:52:00.116 -> OK
13:52:00.116 -> 
0Send successful, no error
13:52:00.116 -> 
13:52:00.116 -> +UUSORD: 0,3
13:52:00.217 -> AT+USORD=0,1

13:52:00.217 -> +USORD: 0,1,"2C"
13:52:00.253 -> 
13:52:00.253 -> OK
13:52:00.253 -> 
,
13:52:00.336 -> +UUSORD: 0,2
13:52:00.336 -> AT+USORD=0,1

13:52:00.336 -> +USORD: 0,1,"30"
13:52:00.336 -> 
13:52:00.336 -> OK
13:52:00.336 -> 0
13:52:00.436 -> +UUSORD: 0,1
[... 352 lines cut ...]
14:47:05.986 -> AT+USORD=0,1

14:47:05.986 -> +USORD: 0,1,"2C"
14:47:05.986 -> 
14:47:05.986 -> OK
14:47:05.986 -> 
,
14:47:06.089 -> +UUSORD: 0,2
14:47:06.089 -> AT+USORD=0,1

14:47:06.089 -> +USORD: 0,1,"30"
14:47:06.124 -> 
14:47:06.124 -> OK
14:47:06.124 -> 
0
14:47:06.193 -> +UUSORD: 0,1
AT+USORD=0,1

14:47:06.228 -> +USORD: 0,1,"5D"
14:47:06.228 -> 
14:47:06.228 -> OK
14:47:06.228 -> ]
14:47:06.503 -> +UUSOCL: 0
14:47:06.503 -> 
14:47:06.503 -> disconnecting.

Rocketct, 2018-10-25:
@Nels52 which version of the library are you using? ok i'll try to reproduce the issue using your suggestions, question are you able to print the _state and _socket value in the GSMClient.cpp?

Nels52, 2018-10-25:
Hi @Rocketct.
I am using the 1.2.1 version of the library with the modification to the GSMClient connect() method described in the first entry of this thread.

Also, I didn't modify the library to print the GSMClient _state variable, however, from the GSMClient code I just figure that if the AT+USOCR=6 hangs the GSMClient state machine is stuck in the CLIENT_STATE_WAIT_CREATE_SOCKET_RESPONSE state.  Similarly, if the AT+USOCO=... command hangs the GSMClient is stuck in the CLIENT_STATE_WAIT_CONNECT_RESPONSE state.

The _socket value is also residual and meaningless after the GSM begin() and GPRS attachGPRS() methods have executed because the modem has been reset which cleans up all socket connections.  In the trace above you will see at 14:47:03.631 where my script issued a GSMClient stop() with the residual _socket value that was no longer valid due to the GSM begin().  This results in a ERROR return because the _socket value is no longer valid.  However, the stop() method at least resets the _socket value to -1 and the ERROR return from AT+USOCL=0 command is harmless.

My basic issue is that I had to make the modification to the GSMClient connect() method because I couldn't see any other way to reset the GSMClient state machine after resetting the modem with a GSM begin().  There may be a better way of resetting the GSMClient state machine, for example, the GSMClient stop() method, however, I just found it was easier with the GSMClient connect() method since it has to be invoked whenever sending a message to the server.

Hope this helps.

Rocketct, 2018-10-26:
Hi @Nels52, yes i understand the error, however i suggest you to update to the 1.3.0 where was implemented many important change that could resolve the problem, could you try this one and let me know if the state of state machine of the client follow to be not updated?

Nels52, 2018-10-26:
Hi @Rocketct.
I will download the latest library and give it a try. It looks like it has a method to retrieve the current client state so I will incorporate that in my script.
Regards

Nels52, 2018-10-26:
Hi @Rocketct 
I installed MKRGSM library 1.3.0 and reproduced the issue by disconnecting the antenna before my script was preparing to send a message to the server.  I seem the same error, i.e. the GSMClient was left in the CLIENT_STATE_WAIT_CONNECT_RESPONSE state after the GSM begin() and GPRS attachGPRS() complete successfully.  My script eventually recovers because it reinitializes GSM and GPRS each time an unexpected result is returned from the GSMClient connect() method call and eventually the GSMClient state is driven back to CLIENT_STATE_CREATE_SOCKET.

I inserted a  Serial print of the GSMClient state just after the following logic in the GSMClient connect() method:

  if (_socket != -1) {
    stop();
  }

**Summary of trace events:**

- Disconnect the antenna.
- At 13:21:28.534 the GSMClient state is 0 (CLIENT_STATE_IDLE) as it should be.
- At 13:22:28.530 AT+USOCO=... hangs because antenna is not connected.
- At 13:22:28.530 my script times out and restarts GSM by calling begin().
- Reconnect the antenna.
- At 13:22:42.065 GSM registers successfully with the network.
- Call GPRS attachGPRS() to initialize the packet network.
- At 13:22:46.578 attached successfully to APN.
- At 13:22:46.612 GSMClient connect() fails because the GSMClient state is 8 (CLIENT_STATE_WAIT_CONNECT_RESPONSE).
- The script restarts GSM again by calling begin().
- At 13:23:02.765 GSM and GPRS complete initialization successfully.
- At 13:23:02.765 GSM connect() fails because GSMClient state is 9 (CLIENT_STATE_CLOSE_SOCKET).
- The script restarts GSM again.
- At 13:23:18.939 GSMClient connect() succeeds because the GSMClient state has progressed to 10 (CLIENT_STATE_WAIT_CLOSE_SOCKET).
- The connect() logic is now able to set the state to CLIENT_STATE_CREATE_SOCKET and the library is able to connect to the server and send its message.

**Debug trace output.:**

`13:21:28.534 -> connecting...
13:21:28.534 -> GSMClient state: 0
AT+USOCR=6

13:21:28.638 -> +USOCR: 0
13:21:28.638 -> 
13:21:28.638 -> OK
AT+USOCO=0,"cloudsocket.hologram.io",9999

13:22:28.530 -> Connection Timed OUT loopCount = 600
13:22:28.530 -> Connection failed
13:22:28.530 -> Restarting Web client
13:22:28.530 -> startWebClient HomeMonitorV4 Build 10-26-2018
AT

13:22:32.689 -> OK
13:22:32.689 -> AT+UPSV=3

13:22:32.689 -> OK
AT+CPIN?

13:22:34.799 -> +CPIN: READY
13:22:34.799 -> 
13:22:34.799 -> OK
AT+CMGF=1

13:22:35.009 -> OK
AT+UDCONF=1,1

13:22:35.215 -> OK
AT+CTZU=1

13:22:35.421 -> OK

13:22:35.491 -> +UMWI: 0,1
13:22:35.491 -> 
13:22:35.491 -> +UMWI: 0,2
13:22:35.491 -> 
13:22:35.491 -> +UMWI: 0,3
13:22:35.491 -> 
13:22:35.491 -> +UMWI: 0,4
[... 648 lines cut ...]
13:23:21.186 -> OK
AT+USORD=0,512

13:23:21.219 -> +USORD: 0,0,""
13:23:21.219 -> 
13:23:21.219 -> OK

13:23:21.324 -> +UUSORD: 0,5
13:23:21.324 -> AT+USORD=0,512

13:23:21.324 -> +USORD: 0,5,"5B302C305D"
13:23:21.324 -> 
OK
13:23:21.358 -> [0Send successful, no error
13:23:21.358 -> ,0]AT+USORD=0,512

13:23:21.598 -> ERROR

13:23:21.633 -> +UUSOCL: 0
13:23:21.633 -> AT+USOCL=0

13:23:21.633 -> ERROR

13:23:21.737 -> disconnecting.
`

Nels52, 2018-10-30:
Hi @Rocketct 
Thanks for looking at this.  I looked at the changes to GSMClient::connect() in the MKRGSM/tree referenced above and I will just incorporate them in the 1.3.0 library I am running.  They should allow me to navigate the GSMClient state machine in asynchronuous mode using the GSMClient connect() and ready() methods.  I agree with the idea of the GSMClient state machine operation (as well as GSM and GPRS state machine operations) being as opaque as possible.

Rocketct, 2018-10-31:
Hi @Nels52 at least the easy solution was update the _state in GSMClient::stop() we made a new PR https://github.com/arduino-libraries/MKRGSM/pull/58 and seems that it works could you test it and let me know if ok?

Nels52, 2018-10-31:
Hi @Rocketct 
Tested with the change to GSMClient::stop() and it worked fine.  Makes sense to put change in stop() since that terminates the socket.

Rocketct, 2018-11-02:
@Nels52 i have closed the issue in favor of https://github.com/arduino-libraries/MKRGSM/pull/58 let me know if ok?

closed 2018-11-02 by Rocketct

Nels52, 2018-11-02:
Hi @Rocketct 
Closing this issue with #58 is fine.  In fact, initializing _state in GSMClient::stop() is a better solution.

Nels52, 2018-11-06:
Hi @Rocketct 
I have run across a situation where the change to GSMClient::stop() fails to address an AT command hang during the connection sequence.  Specifically, if the AT+USOCR= hangs while attempting to create a connection the socket is left at -1 and the GSMClient state is left at CLIENT_STATE_WAIT_CREATE_SOCKET_RESPONSE.  When stop() is called to clean up the failed connection attempt the state is not reset to CLIENT_STATE_IDLE because the socket is -1.  I have moved the _state = CLIENT_STATE_IDLE statement to the top of the stop() method before the socket check and that solves the problem.  I don't believe this has any negative impact on the use of the GSMClient::stop() method.

sandeepmistry, 2018-11-06:
@Nels52 could you please open a pull request for your suggestion above?

Nels52, 2018-11-12:
Hi @Rocketct 
Thanks for your attention to this issue.