arduino-libraries/MKRGSM #57: after the 4th request GSMSSLClient is tilt

compunited, 2018-10-23:
after the 4th request i have this in the debugmode:

OK
AT+USOSEC=4,1

OK
AT+USOCO=4,"iotdatawriter.azurewebsites.net",443

ERROR
AT+USOCL=4

reconnecting doesn't help its always client.ready() ==2

Rocketct, 2018-10-24:
hi @compunited could you share a minimal sketch that reproduce the issue?

compunited, 2019-02-27:
it seems like a proplem with the tcp sockets.  AT+USOCO is counting up from AT+USOCO=0 to AT+USOCO=4 with each request. if it is 4 the next request will fail.

sandeepmistry, 2019-03-05:
@compunited as @Rocketct mentioned in https://github.com/arduino-libraries/MKRGSM/issues/57#issuecomment-432538629 please provide us a sketch to reproduce the issue, then we can re-open this issue.