arduino-libraries/MKRGSM #71: UDP Send timeout is 3Minutes!!

compunited, 2018-12-12:
As you can see the UDP Timeout is 3 Minutes. That's far too long and needs to be changeable. If the message isn't transferred within a few seconds it'll never be transferred successfully.

00:32:37.794 -> AT+USOST=0,"iot.blafasel.de....
00:32:38.918 -> +USOST: 0,112
00:32:38.918 -> 
00:32:38.918 -> OK
00:35:37.853 -> AT+CSQ
00:35:37.853 -> +CSQ: 8,1

compunited, 2018-12-28:
i can't resolve it to a line of code, otherwise i could fix it myself. It happens if there is bad reception. My workarround is to reset it by the watchdog but loosing data

compunited, 2018-12-28:
I don't think the problem is with the library itself. This is probably a setting in the GSM chip, but in the thousand pages of documentation there are plenty of timeouts

sandeepmistry, 2019-01-02:
> I don't think the problem is with the library itself.

@compunited ok, I suggest we follow up with u-blox support on this then. Would you be able to contact them?

compunited, 2019-01-03:
https://forum.u-blox.com/index.php/24280/sara-u201-minute-timeout

victorh800, 2019-06-19:
hi, @compunited. did you get support any support from u-blox?

I have an unrelated issue with the U201. considering the quasi-abandonment of their forum, I have little faith it can provide any meaningful help from u-blox themselves. well, at least some customers do provide some help.