arduino-libraries/MKRGSM #74: ist seems UMTS band is not supported

compunited, 2019-01-05:
according to the ublox doku sara-u201 supports UMTS but not in the arduino definition

bool GSMBand::setBand(String band)
{
  const char* bands;

  if (band == GSM_MODE_EGSM) {
    bands = "900";
  } else if (band == GSM_MODE_DCS) {
    bands = "1800";
  } else if (band == GSM_MODE_PCS) {
    bands = "1900";
  } else if (band == GSM_MODE_EGSM_DCS) {
    bands = "900,1900";
  } else if (band == GSM_MODE_GSM850_PCS) {
    bands = "850,1900";
  } else if (band == GSM_MODE_GSM850_EGSM_DCS_PCS) {
    bands = "800,850,900,1900";
  } else {
    return false;
  }

From the ublox doku ( https://www.u-blox.com/sites/default/files/u-blox-CEL_ATCommands_%28UBX-13002752%29.pdf ):

7.23.3    Defined values
Parameter Type Description
<band_x> Number • 0: restore the factory-programmed configuration of the module
GSM band:
• 900: selection of 900 MHz band
• 1800: selection of 1800 MHz band
• 1900: selection of 1900 MHz band
• 850: selection of 850 MHz band
• 450: selection of 450 MHz band
• 480: selection of 480 MHz band
• 750: selection of 750 MHz band
• 380: selection of 380 MHz band
• 410: selection of 410 MHz band
• 710: selection of 710 MHz band
• 810: selection of 810 MHz band
UMTS band:
• 1: band UMTS I
• 2: band UMTS II
• 3: band UMTS III
• 4: band UMTS IV
• 5: band UMTS V
• 6: band UMTS VI
• 7: band UMTS VII
• 8: band UMTS VIII
• 9: band UMTS IX
• 10: band UMTS X
• 11: band UMTS XI
• 12: band UMTS XII
• 13: band UMTS XIII
• 14: band UMTS XIV
• 15: band UMTS XV
• 16: band UMTS XVI
• 17: band UMTS XVII
• 18: band UMTS XVIII
• 19: band UMTS XIX
• 20: band UMTS XX
• 21: band UMTS XXI
• 22: band UMTS XXII
• 25: band UMTS XXV
UMTS TDD band:
• 201: band UMTS TDD A
• 202: band UMTS TDD B
• 203: band UMTS TDD C
• 204: band UMTS TDD D
• 205: band UMTS TDD E
• 206: band UMTS TDD F

So in my opinion there should be something like 

  } else if (band == GSM_MODE_EGSM_DCS_UMTS) {
    bands = "900,1900,1";

sandeepmistry, 2019-01-07:
Hi @compunited,

Please submit a pull request to clarify your suggested changes.

From the product page: https://www.u-blox.com/en/product/sara-u2-series

The SARA-201 supports quad band GSM and the following UTMS bands: 800, 850, 900, 1900, 2100

compunited, 2019-01-07:
I can't create a pull request with smartgit (normally im working with VS/TFS)
the two files are attached 

[UMTS_changes.zip](https://github.com/arduino-libraries/MKRGSM/files/2733393/UMTS_changes.zip)

compunited, 2019-01-08:
@sandeepmistry you are right. it should be 2100 not 1. I flipped on page  in the docu but nevertheless something like this is missing

} else if (band == GSM_MODE_EGSM_DCS_UMTS) {
bands = "900,1800,2100";

compunited, 2019-01-08:
by the way... take some time and go to this code block from GSMBand.cpp
bool GSMBand::setBand(String band)
{
  const char* bands;

  if (band == GSM_MODE_EGSM) {
    bands = "900";
  } else if (band == GSM_MODE_DCS) {
    bands = "1800"; **Correct**
  } else if (band == GSM_MODE_PCS) {
    bands = "1900"; **Correct**
  } else if (band == GSM_MODE_EGSM_DCS) {
    bands = "900,1900";**WRONG! should be 900,1800**
  } else if (band == GSM_MODE_GSM850_PCS) {
    bands = "850,1900";
  } else if (band == GSM_MODE_GSM850_EGSM_DCS_PCS) {
    bands = "800,850,900,1900"; **WRONG! should be 850,900,1800,1900**
  } else {
    return false;
  }

compunited, 2019-01-16:
@sandeepmistry any updates?

Rocketct, 2019-01-29:
Closed via #78

closed 2019-01-29 by Rocketct