arduino-libraries/MKRGSM #75: GSMClient defect when sending >256 chars

winijenh, 2019-01-11:
The GSMClient library seems to have code to deal with sending messages that have more than 256 chars, by chopping them in 256 byte chucks, but it doesn't function properly. This is part of the current (v1.3.2) code:
`  size_t written = 0;
  String command;

  while (size) {
    size_t chunkSize = size;

    if (chunkSize > 256) {
      chunkSize = 256;
    }

    command.reserve(19 + chunkSize * 2); 

    command += "AT+USOWR=";
    command += _socket;
    command += ",";
    command += chunkSize;
    command += ",\"";

    for (size_t i = 0; i < chunkSize; i++) {
      byte b = buf[i + written];`

The main problem is in this line: 
`command += "AT+USOWR=";`
Durring the 2nd time through the while loop, when attempting to write chars 256 to 511, this line concatenates after the first 256 bytes, writing beyond the reserved String space. It should start from an empty command buffer.

I've modified the code as follows to overcome this:
`  size_t written = 0;
  String command;
  command.reserve(19 + (size>256 ? 256 : size) * 2); // moved from below; do this once

  while (size) {
    size_t chunkSize = size;

    if (chunkSize > 256) {
      chunkSize = 256;
    }

//    command.reserve(19 + chunkSize * 2); // moved to above

    command = "AT+USOWR="; // changed from '+=' to '='
    command += _socket;
    command += ",";
    command += chunkSize;
    command += ",\"";

    for (size_t i = 0; i < chunkSize; i++) {
      byte b = buf[i + written];
`

winijenh, 2019-01-11:
Sample code and results, using the GSMClient example from the library:
`  Serial.println("connecting...");

  // if you get a connection, report back via serial:
  if (client.connect(server, port)) {

    const char *sendbuf = "POST / HTTP/1.1\r\nHost: arduino.cc\r\nContent-Type: application/text\r\nConnection: close\r\nContent-Length: 300\r\n\r\n012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789\r\n\r\n";

    Serial.print("Sending ");
    Serial.print(strlen(sendbuf));
    Serial.println(" bytes");
    
    client.write((uint8_t*)sendbuf,strlen(sendbuf));
`

Relevant output with modem.debug():
`connecting...
AT+USOCR=6

+USOCR: 0

OK
AT+USOCO=0,"example.org",80

OK
Sending 413 bytes
AT+USOWR=0,256,"504F5354202F20485454502F312E310D0A486F73743A2061726475696E6F2E63630D0A436F6E74656E742D547970653A206170706C69636174696F6E2F746578740D0A436F6E6E656374696F6E3A20636C6F73650D0A436F6E74656E742D4C656E6774683A203330300D0A0D0A303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536"

+USOWR: 0,256

OK
AT+USOWR=0,256,"504F5354202F20485454502F312E310D0A486F73743A2061726475696E6F2E63630D0A436F6E74656E742D547970653A206170706C69636174696F6E2F746578740D0A436F6E6E656374696F6E3A20636C6F73650D0A436F6E74656E742D4C656E6774683A203330300D0A0D0A303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536"AT+USOWR=0,157,"3738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738390D0A0D0A"

ERROR
`

Output with library changes as described above:
`connecting...
AT+USOCR=6

+USOCR: 0

OK
AT+USOCO=0,"example.org",80

OK
Sending 413 bytes
AT+USOWR=0,256,"504F5354202F20485454502F312E310D0A486F73743A2061726475696E6F2E63630D0A436F6E74656E742D547970653A206170706C69636174696F6E2F746578740D0A436F6E6E656374696F6E3A20636C6F73650D0A436F6E74656E742D4C656E6774683A203330300D0A0D0A303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536"

+USOWR: 0,256

OK
AT+USOWR=0,157,"3738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738393031323334353637383930313233343536373839303132333435363738390D0A0D0A"

+USOWR: 0,157

OK

<removed a bunch of read commands/responses>

+UUSORD: 0,497
AT+USORD=0,512

+USORD: 0,497,"485454502F312E3120343034204E6F7420466F756E640D0A436F6E74656E742D547970653A20746578742F68746D6C0D0A446174653A204672692C203131204A616E20323031392031323A33353A353120474D540D0A5365727665723A2045435320286463612F32343933290D0A436F6E74656E742D4C656E6774683A203334350D0A436F6E6E656374696F6E3A20636C6F73650D0A0D0A3C3F786D6C2076657273696F6E3D22312E302220656E636F64696E673D2269736F2D383835392D31223F3E0A3C21444F43545950452068746D6C205055424C494320222D2F2F5733432F2F445444205848544D4C20312E30205472616E736974696F6E616C2F2F454E220A20202020202020202022687474703A2F2F7777772E77332E6F72672F54522F7868746D6C312F4454442F7868746D6C312D7472616E736974696F6E616C2E647464223E0A3C68746D6C20786D6C6E733D22687474703A2F2F7777772E77332E6F72672F313939392F7868746D6C2220786D6C3A6C616E673D22656E22206C616E673D22656E223E0A093C686561643E0A09093C7469746C653E343034202D204E6F7420466F756E643C2F7469746C653E0A093C2F686561643E0A093C626F64793E0A09093C68313E343034202D204E6F7420466F756E643C2F68313E0A093C2F626F64793E0A3C2F68746D6C3E0A"

OK
HTTP/1.1 404 Not Found
Content-Type: text/html
Date: Fri, 11 Jan 2019 12:35:51 GMT
Server: ECS (dca/2493)
Content-Length: 345
Connection: close

<?xml version="1.0" encoding="iso-8859-1"?>
<!DOCTYPE html PUBLIC "-//W3C//DTD XHTML 1.0 Transitional//EN"
         "http://www.w3.org/TR/xhtml1/DTD/xhtml1-transitional.dtd">
<html xmlns="http://www.w3.org/1999/xhtml" xml:lang="en" lang="en">
	<head>
		<title>404 - Not Found</title>
	</head>
	<body>
		<h1>404 - Not Found</h1>
	</body>
</html>
AT+USORD=0,512

ERROR

+UUSOCL: 0
AT+USOCL=0

ERROR

disconnecting.
`

sandeepmistry, 2019-01-11:
Hi @winijenh,

Your suggestion in https://github.com/arduino-libraries/MKRGSM/issues/75#issue-398252368 looks good to me, would you like to open a pull request with the change?

winijenh, 2019-01-11:
Sorry, I have no experience with github, don't know how to do that,

sandeepmistry, 2019-01-14:
Hi @winijenh,

I've created a pull request: https://github.com/arduino-libraries/MKRGSM/pull/76, can you take a quick look at the changes to make sure they are as you suggested. Thanks.

sandeepmistry, 2019-01-18:
Closed via https://github.com/arduino-libraries/MKRGSM/pull/76

closed 2019-01-18 by sandeepmistry