arduino-libraries/MKRGSM #79: MKR GSM hangs after few loop runs

Basi0815, 2019-02-07:
Hello,
i am using an mkr gsm 1400 for sending an analog value to ThingSpeak. This is used for Checking the waterlevel in an small hydrostatic power station. The mobile connection at the plant ist not great but works for a Ardunio uno an Sim900 module for one week. After a few runs to the programm loop stops after saing "Channel Update erfolgreich". After that no data is send anymore. The Programm is out of the ThingSpeak libary, and was upgraded for my purposes. (sending sms when level drops under an defined level, modem reset after 10 failed data sends)
I am using vodafone, in Germany.
Hope anybody can help me. This is my project for my bachelor thesis. :-)

//Einbindung der erforderlichen Bibliotheken 
#include "ThingSpeak.h"
#include <MKRGSM.h>
#include "secrets.h"
#include "core_cm0plus.h"

// SIM PIN Number
const char PINNUMBER[]     = SECRET_PIN;
// APN data
const char GPRS_APN[]      = SECRET_GPRS_APN;
const char GPRS_LOGIN[]    = SECRET_GPRS_LOGIN;
const char GPRS_PASSWORD[] = SECRET_GPRS_PASS;

GSMClient client;
GPRS gprs;
GSM gsmAccess;
GSM_SMS sms;

unsigned long myChannelNumber = SECRET_CH_ID;
const char * myWriteAPIKey = SECRET_WRITE_APIKEY;

//Variablen initialisieren
float hmess = 0.0;
float pegel = 0.0;
float fuellstand = 0.0;
int errorcheck = 0;
int pegelkleiner0_5m = 0;

//Setup wird nur einmal durchlaufen
void setup() {
  
  //Debug Modus für Mobilfunk Modem
  MODEM.debug();

    Serial.begin(57600);  //serielle Kommunikation initialisieren 

    Serial.println("Starting Arduino web client.");
    boolean connected = false;
    
     //10 Sekunden auf Verbindung warten 
    delay(30000);
      
      while (!connected) {
        if ((gsmAccess.begin(PINNUMBER) == GSM_READY) &&
          (gprs.attachGPRS(GPRS_APN, GPRS_LOGIN, GPRS_PASSWORD) == GPRS_READY)) {
          connected = true;
        } else {
          Serial.println("Nicht Verbunden!");
          delay(5000);
        }
      }
    Serial.println("Verbunden!");
    ThingSpeak.begin(client);  //ThingSpeak initialisieren
}

void loop() {
 
    //Variablen
    float h0 = 3.95;
    //Messfenster 0,35m-5,0m -> 4,65m
    //Umrechnungsfaktor 4,65m/4095
    float faktor = (4.65/4095);

    //ADC auf 12 Bit Genauigkeit
    analogReadResolution(12);
    //Analogeingang in Integerwert einlesen
    int adc = analogRead(A0);
    //0-5m aus 0-3.3V Eingangssignal
    hmess = ((adc*faktor)+0.35);  
    //Messhoehe von der absoulten Sensorhoehe abziehen->Wasserstand 
    pegel = (h0 - hmess); 

    //SMS bei vorgegebenen Pegel 0,5m enstspricht einem hmess von 3,1m
    //adc=3,1m/(4,65/4095)=2730
    if(adc>2730 && pegelkleiner0_5m == 0){
    char txtMsg []= "Achtung Pegel unter 0,5m!";
    char remoteNum[] = "0160***";
    sms.beginSMS(remoteNum);
    sms.print(txtMsg);
    sms.endSMS();
    pegelkleiner0_5m = 1;
    }

    //SMS versand wieder zulassen wenn Pegel einmal groesser 0,5m war
    if(adc<2730){
      pegelkleiner0_5m = 0; 
      }

    //Ausgabe der berechneten Werte
    Serial.println("Berechneter Pegel:");
    Serial.println(pegel);
    fuellstand = ((pegel/3.40)*100);
    Serial.println("Berechneter Fuellstand [%]:");
    Serial.println(fuellstand);

  // Setzte Channel Felder mit Werten
  ThingSpeak.setField(1, pegel);
  ThingSpeak.setField(2, fuellstand);
    
  // Schreiben von Daten in ThingSpeaks Channel
  int x = ThingSpeak.writeFields(myChannelNumber, myWriteAPIKey);
  
  if(x == 200){
    Serial.println("Channel Update erfolgreich.");
    errorcheck = 0;
  }
  else{
    Serial.println("Problem beim updaten des Channels. HTTP error code " + String(x));
    errorcheck = (errorcheck + 1);
    Serial.println("Fehleranzahl in Folge: ");
    Serial.println(errorcheck);
  }
  if(errorcheck == 10){
    //Nach 10 Uebertragungsfehlern in Folge fuere Reset aus
    Serial.println("10 Fehler in Folge beim Uebertragen -> Software Reset");
    errorcheck = 0;
    delay(200);
    //Software Resetfunktion
    NVIC_SystemReset();
    }

  delay(45000); //45 Sekunden warten bis zum neuen Update des Channels
}

/*
 * ERRORCODES aus ThingSpeak.h
 * 
#define OK_SUCCESS              200     // OK / Success
#define ERR_BADAPIKEY           400     // Incorrect API key (or invalid ThingSpeak server address)
#define ERR_BADURL              404     // Incorrect API key (or invalid ThingSpeak server address)
#define ERR_OUT_OF_RANGE        -101    // Value is out of range or string is too long (> 255 bytes)
#define ERR_INVALID_FIELD_NUM   -201    // Invalid field number specified
#define ERR_SETFIELD_NOT_CALLED -210    // setField() was not called before writeFields()
#define ERR_CONNECT_FAILED      -301    // Failed to connect to ThingSpeak
#define ERR_UNEXPECTED_FAIL     -302    // Unexpected failure during write to ThingSpeak
#define ERR_BAD_RESPONSE        -303    // Unable to parse response
#define ERR_TIMEOUT             -304    // Timeout waiting for server to respond
#define ERR_NOT_INSERTED        -401    // Point was not inserted (most probable cause is the rate limit of once every 15 seconds)
 */

Rocketct, 2019-02-20:
Hi @Basi0815 i'm testing your sketch and seems work it run until yesterday, could you enable the debugger by  setting the GSM gsmAccess(true) and share the results?

Basi0815, 2019-02-20:
Hi @Rocketct i have tested the code now from my office and you are right it is working fine.  The only difference is that i am using a sim card from a different carrier.(German carrier Congstar, using Telekom network).
Is this maybe a problem caused by the bad Internet connection or the carrier at the power plant? 

Unfortunately, I can not provide a log until next week. Then I'm back on the spot.

Rocketct, 2019-02-21:
Yes the board is really sensitive to the power plant, and also to the quality of the signal. You can check some data about the connection quality by using the  AT command ```AT+CIND? ``` from the a t command manual https://www.u-blox.com/sites/default/files/u-blox-CEL_ATCommands_(UBX-13002752).pdf pag 46,  you can send it using the SerialGSMPassthroug sketch.

let me know the results.

Basi0815, 2019-02-26:
Hi here is the result form the plant:

`AT+CIND?

09:22:47.068 -> +CIND: 5,0,65535,0,0,0,65535,0,65535,0,0,1
09:22:47.068 -> 
09:22:47.068 -> OK
AT+CIND?

09:22:54.751 -> +CIND: 5,0,65535,0,0,0,65535,0,65535,0,0,1
09:22:54.751 -> 
09:22:54.751 -> OK
`

results at my office:
`AT+CIND?

11:05:48.203 -> +CIND: 5,3,1,0,1,0,0,0,2,0,0,1
11:05:48.203 -> 
11:05:48.203 -> OK
`

Rocketct, 2019-02-28:
The second and the third parameters are releted to signal and service, from the manual:
```"signal": signal quality. See mapping in the note below:
- 0  (< -105 dBm or unknown)
- 1  (< -93 dBm)
- 2  (< -81 dBm)
- 3   (< -69 dBm)
- 4   (< -57 dBm)
- 5   (>= -57 dBm)

"service": network service availabilityo0: not registered to any networko1: registered to the networko65535: indication not available```

 have you also a log of the connection procedure? this is vailable by setting 
```GSM gsmAccess(true);```

Basi0815, 2019-02-28:
Connection time is ok. I going now with the Arduino UNO and SIM900 modul. The MKR is dont work for me. For this price the mkr is very disappointing.

compunited, 2019-05-05:
@basi0815 i opened the same bugs 6 month ago and it was closed... 
https://github.com/arduino-libraries/MKRGSM/issues/57