arduino-libraries/MKRGSM #84: Class GSM and other use private variables instead of protected

johncpang, 2019-03-31:
I'm looking for a reusable-way to use the library in async.  Started with the `GsmWebClient` example, I tried to rewrite the first call `GSM::begin(PINNUMBER)` into synchronized function.  My attempt is to sub-class `GSM` as `AsyncGSM`, and send `synchronous=false` to `begin()`.  The overloaded function will take a callback (pointer-to-function) so that when `begin()` finished (whether success or failed), it will callback the caller.

To my understanding, I should rewrite the following code from `GSM::begin()` in my overloaded function, and run the code via main `loop()`.

      unsigned long start = millis();

      while (ready() == 0) {
        if (_timeout && !((millis() - start) < _timeout)) {
          _state = ERROR;
          break;
        }

        delay(100);
      }

However, I cannot access `_timeout` as it is a private variable instead of protected.  I believe, the whole library should consider using `protected` over `private`, or provide both getters and setters, so that we can extend the library by sub-classing.

sandeepmistry, 2019-04-01:
Hi @johncpang,

Adding a getter for it sounds good, would you be able to submit a pull request for this?

johncpang, 2019-04-02:
Hi @sandeepmistry, I'd like to do that.  When I've got the library works in async, then I'll make a pull request.

I'm able to make class `GSM` works aysnc.  But `MODEM.begin()` is a synchronized function and it takes 10 seconds to completed.  Can class `MODEM` works in async mode?  Is there an async-version `MODEM`?

As a test or indicator, I blink the onboard LED using a Timer.  It stops blinking when `MODEM.begin()` called, and resumes when `MODEM.begin()` returned.

johncpang, 2019-04-30:
@sandeepmistry Hi, I've submitted a pull request for this.  Please help and merge it.