arduino-libraries/MKRGSM #92: Random hang of MKR GSM 1400

Freshrik, 2019-05-21:
Hi!

I see some related topics but none without GSM communication running. I am having issues with the MKR GSM 1400 hanging at random times just as other have. Some have related te issue to a hang in a while loop within communication. I howerver, only use pin change interrupts, a timer and toggle an LED. I am yet to set up communication. There seems to be an issue even without the GSM-communication.

Thanks you for any input!

```
#include "timer.h"

#define OUTPIN 6
#define RESETPIN 3
#define LEDPIN2 1
#define LEDPIN1 0

//Timer object
Timer timer;
//int conv;
volatile int pulses[21];
int pos;
//int WaitForEnd;
//volatile int count;
//volatile int cUp;
//volatile int cDn;
volatile int elapsed;
volatile int analyze;
int newError;
int oldError;
int error;
int led;

void setup()
{
//  Serial.begin(1200);

  pinMode(RESETPIN, INPUT_PULLUP);
  pinMode(LEDPIN2, INPUT);
  pinMode(LEDPIN1, INPUT);
  pinMode(OUTPIN, OUTPUT);
  //  conv = 0;

  attachInterrupt(digitalPinToInterrupt(LEDPIN1), triggerUp, RISING);
  attachInterrupt(digitalPinToInterrupt(LEDPIN2), triggerDown, FALLING);

  for (int i = 0; i < 21; i++) {
    pulses[i] = 0;
  }
  pos = 0;
  //count = 0;
  //cUp = 0;
  //cDn = 0;

  oldError = -1;
  error = -1;
  
  //Start the timer
  timer.start();

  // OLD SETUP
  //attachInterrupt(digitalPinToInterrupt(STARTPIN), SayHi, RISING);
  //attachInterrupt(digitalPinToInterrupt(STOPPIN), SayBye, FALLING);

}

void loop() {

  if (analyze == 1) {
//    Serial.print("pulses = [");
//    for ( int i=0 ; i < pos-1 ; i++ ){
//      Serial.print(pulses[i]);
//      if (i<pos-2){
//        Serial.print(",");
//      }
//    }
//    Serial.println("]");
    if (pulses[pos-1] > 9000 && pos == 2){
      newError = 0;
    } else {
      newError = pos/2;
    }
    if (newError == oldError){
      error = newError;
    } else {
      oldError = newError;
      error = 12;
    }
    for (int i=0 ; i < 21 ; i++){
      pulses[i]=0;
    }

    pos = 0;
    analyze = 0;
    if (led ==1) {
      led = 0;
      digitalWrite(OUTPIN,LOW);
    } else {
      led = 1;
      digitalWrite(OUTPIN,HIGH);
    }
  }

  
  if (timer.getElapsedTime()>=20000 && digitalRead(LEDPIN1)==HIGH){
    timer.reset();
    error = -1;
    analyze = 1;
  }
  
  //Update the timer (NEEDED!)
  noInterrupts();
  timer.update();
  interrupts();
}

void reset() {
  //cUp = 0;
  //cDn = 0;
  //count = 0;
}

void triggerUp() {
  noInterrupts();
  delay(1);
  if (digitalRead(LEDPIN1) == HIGH && analyze == 0 && pos != 0) {
    elapsed = timer.getElapsedTime();
    timer.reset();
    //Serial.print("ms sedan senaste nedflanken: ");
    //Serial.println(elapsed);
    pulses[pos] = elapsed;
    pos++;
    if (elapsed > 500) {
      analyze = 1;
    }
    //count++;
    //Serial.println(count);
    //cUp++;
    //Serial.print("Upp: ");
    //Serial.println(cUp);
  }
  interrupts();
}
void triggerDown() {
  noInterrupts();
  delay(1);
  if (digitalRead(LEDPIN2) == LOW) {
    elapsed = timer.getElapsedTime();
    timer.reset();
    //Serial.print("ms sedan senaste uppflanken: ");
    //Serial.println(elapsed);
    pulses[pos] = elapsed;
    pos++;
    //count++;
    //Serial.println(count);
    //cDn++;
    //Serial.print("Ner: ");
    //Serial.println(cDn);
  }
  interrupts();
}
```

Rocketct, 2019-06-20:
hi @Freshrik which is the timer library that you are using?

Rocketct, 2019-09-06:
Closed due to lack of feedback @Freshrik please reopen if you need

closed 2019-09-06 by Rocketct