arduino-libraries/MKRGSM #94: GSMClient.connect loops forever

adampulley, 2019-05-27:
I had my new code working perfectly on test overnight, but after several hours on soak test the GSMClient.connect() part of it failed and ever since it will never connect again.  Attached is a capture showing the issue all the way from startup with debugging on (there are a few of my console output messages amongst the gsm debug output also).  The http GET request is sent but the reply never comes.  I've tested with my own server as the target and the http request never actually makes it to the web server.  I can't see any issue with the SIM card and there's no network operator message been recevied indicating any issue rom them.  I can send a SMS to it and have the mkrgsm code get it no problem still, just GSMClient.connect() fails to work properly

[issue capture.txt](https://github.com/arduino-libraries/MKRGSM/files/3224811/issue.capture.txt)

Rocketct, 2019-06-06:
Hi @adampulley could you share a minimal sketch to reproduce the issue?

adampulley, 2019-06-09:
Hi sorry its taken a few days to do but I've chased this down to an issue with the O2 network, the mkrgsm code has turned out not to be the cause

closed 2019-06-09 by adampulley