arduino-libraries/MKRGSM #98: Iterating on incorrect PIN is default behavior, locks SIM

8bitkick, 2019-07-10:
GSM.begin(PIN) is called in a loop in all example code. Therefore users omitting or setting an incorrect PIN results in SIM card being locked when the sketch is executed, as it keeps trying.

Might be better handled by catching the AT response for incorrect PIN on first attempt, halting and displaying an error message so the user can correct the PIN.

sandeepmistry, 2019-07-15:
So the current enum for the return value of `GSM::begin(...)` is limited to:
```
enum GSM3_NetworkStatus_t { ERROR, IDLE, CONNECTING, GSM_READY, GPRS_READY, TRANSPARENT_CONNECTED, GSM_OFF};
```

Right now, `ERROR` is returned for the following scenarios:
* any AT command error result, including invalid pin
* network registration denied

@cmaglie @facchinm @tigoe should we consider to add a new enum value specifically for invalid pin? Another option (which is more backwards compatible), is to add another method for error reason.

Any thoughts on this?

tigoe, 2019-07-19:
I'd go with the backwards-compatible version.

roener, 2020-07-09:
I had the same problem and solved it for me by asking for remaining pin attempts.
Maybe this is helpful for anybody.

```
// custom command
// see page 195 https://www.u-blox.com/sites/default/files/u-blox-CEL_ATCommands_%28UBX-13002752%29.pdf
int getRemainingPinAttempts() {
  int responseStatus;
  String response;
  int remainingAttempts;

  MODEM.send("AT+UPINCNT");
  responseStatus = MODEM.waitForResponse(1000, &response);

  if (responseStatus != 1) {
    // -1 -> timeout
    // 1 -> OK
    // 2 -> ERROR
    // 3 -> NO CARRIER
    return -1;
  }

  int firstSpaceIndex = response.indexOf(' ');
  int firstCommaIndex = response.indexOf(',');

  if (firstSpaceIndex != -1 && firstCommaIndex != -1) {
    String responseSubstring;
    responseSubstring = response.substring(firstSpaceIndex + 1, firstCommaIndex);
    remainingAttempts = responseSubstring.toInt();
    if (remainingAttempts == 0 || responseSubstring == "") {
      remainingAttempts = -1;
    }
  } else {  // error
    remainingAttempts = -1;
  }

  return remainingAttempts;
}
```