arduino-libraries/MKRGSM #101: MKRGSM bug with accents

zolanet, 2019-09-18:
Hello,
I will be using an Arduino MKR GSM 1400 on an installation that is in french.
I have come across a bug in the library where accents are not sent unless I use emojis:

using the receive SMS example https://www.arduino.cc/en/Tutorial/MKRGSMExamplesReceiveSMS 

If I send the following message: 
"Teste avec accès aux accents"

The Serial Monitor returns:
"Teste avec accs aux accents"

But if I send:
"Teste avec accès aux accents 😜"

The serial Monitor returns the following hex values:
"0054006500730074006500200061007600650063002000610063006300E80073002000610075007800200061006300630065006E007400730020D83DDE1C"

Which I can decode to:
"Teste avec accès aux accents 😜"

Is there a way to force hex values to be returned instead of text? 
It's seems that it's the only way to get accents to show up...

Regards 
Marc

zolanet, 2019-09-26:
Following up. The following sentence:
"J'ai décidé d'aller au café, mais il est rempli à craquer. Ça demande de la patience. Il y a des gens qui attendent dans l'entrée! C'est complètement fou!"
returns:
"J'ai dcid d'aller au caf, mais il est rempli craquer. a demande de la patience. Il y a des gens qui attendent dans l'entre! C'est compltement fou!"
But, if I use a "ê":
"J'ai décidé d'aller au café, mais il est rempli à craquer. Il faut être patient. Il y a des gens qui attendent dans l'entrée! C'est complètement fou!"
It returns:
"004A0027006100690020006400E900630069006400E90020006400270061006C006C00650072002000610075002000630061006600E9002C0020006D00610069007300200069006C0020006500730074002000720065006D0070006C0069002000E000200063007200610071007500650072002E00200049006C00200066006100750074002000EA007400720065002000700061007400690065006E0074002E00200049006C00200079002000610020006400650073002000670065006E0073002000710075006900200061007400740065006E00640065006E0074002000640061006E00730020006C00270065006E0074007200E90065002100200043002700650073007400200063006F006D0070006C00E800740065006D0065006E007400200066006F00750021"
Which decodes to:
"J'ai décidé d'aller au café, mais il est rempli à craquer. Il faut être patient. Il y a des gens qui attendent dans l'entrée! C'est complètement fou!"

So it seems that ê and emojis force hex output. Please let me know how to force the arduino to output Hex values instead of characters

regards

zolanet, 2019-09-26:
Going further, the following accented letters return as HEX values
À Â â È Ê ê Ë ë Ï ï Î î Ô ô Ù Û û

The following accented characters are completely deleted from the sting if used by themselves:
à è É é Ü ü ù Ç ç

Rocketct, 2019-09-27:
Hi  @zolanet like replied here https://github.com/arduino-libraries/MKRGSM/issues/31 for the special character, we doesn't support the extended character set because the read() function use a buffer of bytes, you could implement a logic that handles the unicod code points from the _incomingBuffer in the GSM SMS class, and recognize the characters that you need.

zolanet, 2019-09-27:
Thanx for the reply Riccardo,
This is a bit beyond my skills. Is it as simple as:

```
int GSM_SMS::read()
{
  int bufferLength = _incomingBuffer.length();

  if (_smsDataIndex < bufferLength && _smsDataIndex <= _smsDataEndIndex) 
  {
		switch(_smsDataIndex)
		{
			case "é":
				return "00E9"
				_smsDataIndex++
				break;
			default:
				return _incomingBuffer[_smsDataIndex++]
				break;
		}
  }
  return -1;
}
```

if not, can you give a simple example.

regards

salvq, 2019-12-25:
I would also welcome help here...

@zolanet , have you found out the way to convert the special characters like you suggested above ?

I am trying to convert special characters to ASCII to get human readable message rather than HEX code.

Thanks

Rocketct, 2020-03-06:
sorry for late delay, @zolanet  you should check the value in incomig buffer like:
....
if ( _incomingBuffer[_smsDataIndex++] == special_Char_code ) {
   return teh valueyou need
}
return ( _incomingBuffer[_smsDataIndex++]
....

let me find a working solution i'll post as soon i'll find

Rocketct, 2020-03-09:
@zolanet i make some test i got the following results:

- with the actual configuration the character set by the modem allow the to receive only ** GSM default alphabet (3GPP TS 23.038)** 
this means that different from what i report before, is the modem that filter the received message;

- to allow the reception of the special character defined in the standard **(USO/IEC10646) **, you have to:

1. change the character set, add in the **GSM_SMS::available()** in the very begin of the function; the line `MODEM.send("AT+CSCS=\"UCS2\"");` this will changes how the modem message body, 
2. read the value from the incoming buffer convert in the char you want; 

practically the one written above means:

```
while ((c = sms.read()) != -1) {
      //Serial.print((char)c);
      //outbuf += String(c);
      counter++;
      if(c == 48){
        outbuf +='0';
      }
      if(c == 69){
        outbuf +='E';
      }
      if(c == 56){
        outbuf +='8';
      }
     
    }
    if (outbuf == "00E8") {
      Serial.println("è");
    }

    Serial.println("\nEND OF MESSAGE");

    // Delete message from modem memory
    sms.flush();
    Serial.println("MESSAGE DELETED");
  }
```

the output enabling the at debug will be:

without changing the set char:
AT+CMGL="REC UNREAD"

+CMGL: 2,"REC UNREAD","11111111",,"20/03/09,14:08:31+04"

changing the set char:
AT+CSCS="UCS2"
OK
AT+CMGL="REC UNREAD"
+CMGL: 2,"REC UNREAD","00310031003100310031003100310031",,"20/03/09,14:27:48+04"
00E8

pay attention how "11111111" become "00310031003100310031003100310031" (removed my number, this an example i have check the duality and it works) where 0031=1 in the standard **ISO-8859-1** notice that the difference between the first and the seconds read is the presence of  00E8 that is 'é' ,

the example i had used is the following :

```

// include the GSM library
#include <MKRGSM.h>

#include "arduino_secrets.h"
// Please enter your sensitive data in the Secret tab or arduino_secrets.h
// PIN Number
const char PINNUMBER[] = SECRET_PINNUMBER;

// initialize the library instances
GSM gsmAccess(true);
GSM_SMS sms;

// Array to hold the number a SMS is retreived from
char senderNumber[80];

void setup() {
  // initialize serial communications and wait for port to open:
  Serial.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }

  Serial.println("SMS Messages Receiver");

  // connection state
  bool connected = false;

  // Start GSM connection
  while (!connected) {
    if (gsmAccess.begin(PINNUMBER) == GSM_READY) {
      connected = true;
    } else {
      Serial.println("Not connected");
      delay(1000);
    }
  }

  Serial.println("GSM initialized");
  Serial.println("Waiting for messages");
}

void loop() {
  int c;

  // If there are any SMSs available()
  if (sms.available()) {
    Serial.println("Message received from:");

    // Get remote number
    sms.remoteNumber(senderNumber, 80);
    Serial.println(senderNumber);

    // An example of message disposal
    // Any messages starting with # should be discarded
    if (sms.peek() == '#') {
      Serial.println("Discarded SMS");
      sms.flush();
    }
//----------------------- made the string and find the equivalent char value ------------------------
    String outbuf = "";
    // Read message bytes and print them
    while ((c = sms.read()) != -1) {
      //Serial.print((char)c);
      //outbuf += String(c);
      if(c == 48){
        outbuf +='0';
      }
      if(c == 69){
        outbuf +='E';
      }
      if(c == 56){
        outbuf +='8';
      }
     
    }
    if (outbuf == "00E8") {
      Serial.println("è");
    }
//--------------------------------------------------------------------------------------------------------------------

    Serial.println("\nEND OF MESSAGE");

    // Delete message from modem memory
    sms.flush();
    Serial.println("MESSAGE DELETED");
  }

  delay(1000);

}
```

Rocketct, 2020-03-09:
hope this help

Wortoll, 2020-07-14:
Hi there,
I have the same problem. I try to make a sms-printer (with the Adafruit thermal printer)

- é   does not print
- emoticons make the whole message unreadable

Is there a way to receive the sms's always in the same format and then do a translation on that? 
(this way I could change the not compatible characters to a symbol or question mark) ?
maybe first making a string of the whole message?

I tried to understand what you all posted on this topic but I don't seem to get it... 
Hopefully someone knows a solution, this seems as a common problem when receiving sms's

Wortoll, 2020-07-26:
New issue arises
- When using `MODEM.send("AT+CSCS=\"UCS2\"");`  the modem hangs when sending a message that is 2 messages long.
In normal usage, this returns 2 messages

with debug mode enabled it returns 
```
.AT+CSCS="UCS2"

OK
AT+CMGL="REC UNREAD"
```
and then hangs..

if I then reconfigure the modem to print in default (non UCS2) , everything works again...

Could it be a variable that overflows?