arduino-libraries/MKRGSM #102: How to send using GSM character set

adam-pulley, 2019-11-08:
Is it possible to send using the GSM character set?  It doesn't appear to be documented on arduino.cc/en/Reference/MKRGSM and I've tried every possible approach I can think of without success.  It appears MKRGSM is converting some characters to the GSM character set, for instance '@' and '$' sends correctly even though they are characters 0x00 and 0x02 in the GSM character set, but I can't find any way to send the british pound (0x01), euro (0xE2 0x82 0xAC) or general supported accented european characters for instance.  My suspicion is that the MKRGSM is performing a conversion from standard ASCII for some characters, such as @ and $, but not for others and I can't find any way to tell it to or to accept hex character codes with me doing the conversion for it?

adam-pulley, 2019-11-08:
I've just found that if I send a message with special characters it gets output by MKRGSM to me as a string like this: 00480065006C006C006F0020005B005D007B007D00230025005E002A002B003D005F005C007C007E003C003E20AC002400A52022002E002C003F00210027
So it seems there is handling of hex code based messages, does anyone know if this is documetned anywhere, how to determine when the message is hex based and how to send a hex based message?

y-fedorov, 2019-11-11:
It's SMS PDU Format - https://www.activexperts.com/sms-component/sms/pdu/ it can be decoded with diffrernt libraries.

Rocketct, 2020-03-09:
@adam-pulley you can also check on the at  command manual of the sara u201, **be careful we not support as arduino any change or problem made by using that command alone**, say this you can check the section related the AT+CSCS, use to set the character set and an lsit of any the format supported by the sara u201, as suggested by @y-fedorov  could be a good idea try to search some library that that decode it for you, due to the fact we as arduino not support this last.

ogellein, 2020-12-08:
I had simular problem with sending Norwegian letters.
With the MKR GSM 1400 I am able to change the modem encoding.
After: gsmAccess.begin(PINNUMBER);
I add: encoding MODEM.send("AT+CSCS=\"8859-1\"");  //you can also use GSM

And then the input also must match this encoding, i use this one:
String filterInsertSpecialChar(String str){
    for (int i=0; i <= str.length(); i++){
       str.replace("æ",String(char(230))); 
       str.replace("Æ",String(char(198))); 
       str.replace("ø",String(char(248))); 
       str.replace("Ø",String(char(216)));
       str.replace("å",String(char(229))); 
       str.replace("Å",String(char(197))); 
     
    }
    return str;
}
Reference: https://www.u-blox.com/sites/default/files/u-blox-CEL_ATCommands_%28UBX-13002752%29.pdf

eirikora, 2021-06-08:
Thanks @ogellein, your advice worked for me and I am now able to send Norwegian SMSes with the MKR GSM 1400 after having tried some way too complex alleys earlier. 
Have posted an updated code example here that could also be adapted for other European languages: https://forum.arduino.cc/t/how-to-send-sms-with-nordic-characters-on-arduino-mkr-gsm-1400/870836