arduino-libraries/MKRGSM #109: GSM.begin() and GSM_SMS.available() without freezing the loop()?

srv-config, 2020-04-27:
Hi,

The main flaw I experience with MKRGSM lib is inability to do range of things without temporary freezing or slowing down the loop timers.

E.g. each time I attempt to connect to GSM via GSM.begin() the loop() holds on until it gets connected. I tried using async mode and check for GSM.ready() but it was giving same results, with the difference I could check for ready() whenever I liked to.

I tried to combine the connection process with timers but result is still same. Exactly same problem is with GSM_SMS.available(). Each time the function is executed the entire loop is being stopped for ~300-500ms in total what has huge impact on all other processes.

Question is: is it possible to make the two function not handing the loop()?

srv-config, 2020-04-29:
Any solution for this? with such delays the MKR1400 GSM is not usable for serious projects.

CptHolzschnauz, 2020-08-02:
Hi

gsm.begin() comes to the void setup() section and attaching the network needs some time. Use gsmAccess(true) to see the modem messages over the serial monitor. With the connection set up,  SMS.available() in the loop needs some millies.

Cheers