arduino-libraries/MKRGSM #132: Add a way to include more root certs to be used

sergiocastrogom, 2021-04-19:
Hi team!

I have been facing issues using the GsmSSLWebClient example to connect to some servers via HTTPS. I suspect that the cause of my connectivity issue could be related to the fact that the certs used by my server is not listed in https://github.com/arduino-libraries/MKRGSM/blob/master/src/utility/GSMRootCerts.h. This list does include "DigiCert_High_Assurance_EV_Root_CA", however the server I am trying to connect to (webhooks.mongodb-realm.com) uses "DigiCert Global Root CA".

Is there a way for me to manually add such cert? if not, would it be possible to add such functionality?

Thanks a lot in advance for your help.

tommirintala, 2021-11-26:
I have the same problem -- but I would need to add LetsEncrypt -root certificate.

Hugocourr, 2022-04-11:
i made a tutorial for that here: https://github.com/arduino-libraries/MKRGSM/issues/151 here you go :) have fun!