arduino-libraries/MKRGSM #156: TLS handshake needs SNI

bseib, 2022-07-29:
I am having an issue establishing an HTTPS connection to a server on digital ocean. I worked with their engineers to pinpoint what is happening and it turns out that their CloudFlare load balancer is not receiving a Server Name Indication (SNI) during the TLS handshake, and so it refuses the connection.

https://www.cloudflare.com/learning/ssl/what-is-sni/

Can anyone help point me to the right place in the code where this would be added or fixed? I'd be happy to work on a PR with a little help.

manchoz, 2022-07-29:
Hi @bseib, please, look at #88.

bseib, 2022-07-29:
Ah, bummer. :-(

Looks like there might be some firmware update out there somewhere to support SNI now: https://usermanual.wiki/m/d5a29fa230f7ac9362ba2c842b6df3e2c4c29d8ffbb42fe93ee01787a22b1823.pdf

But at first look, I didn't find find their new firmware:  https://www.u-blox.com/en/product-resources?query=u-blox&file_category=Firmware%2520Update