arduino-libraries/MKRGSM #169: Challenges with HTTPS Requests and SSL Certificate Handling on Arduino MKR GSM 1400

ShahabYahoo, 2024-03-28:
Hello!

I've been encountering challenges with my Arduino MKR GSM 1400 module, specifically when making secure HTTPS requests to the OpenCellID API and managing SSL certificates for secure connections. Here are the key issues I've faced:

HTTPS Requests to OpenCellID: Attempting HTTPS GET requests to https://opencellid.org/ consistently results in errors like (AT+USOCO=0,"[opencellid.org](http://opencellid.org/)",443 ERROR ), indicating a failure to establish a connection over port 443. Despite this, I have tested the URL containing the API key manually in a web browser to confirm the correctness of the API key and parameters, which worked as expected, returning the correct data.
Handling SSL Certificates: Despite using the GSMSSLClient library intended for secure connections, I run into (AT+USECMNG=2,0,"AmazonRootCA1" ERROR ), which suggests problems with SSL certificate management on the module. This is concerning since the module is supposed to support TLS 1.2 for secure communications.
Library and Command Conflicts: Efforts to modify the GSMSSLClient library and directly send AT commands have not resolved the issues, hinting at possibly deeper problems with the modem firmware or library compatibility with secure protocols.
I have shared my source code and serial monitor outputs to provide more context on these issues. If anyone in the community has faced similar challenges or has insights on resolving them, I would greatly appreciate your advice. I'm interested in any workarounds, firmware updates, or alternative methods for securely fetching data from web APIs using the MKR GSM 1400.

Your support and suggestions would be invaluable to me as I navigate these challenges. Thank you for taking the time to read and respond.

Source code:
`#include <MKRGSM.h>
#include <ArduinoJson.h>

//  GPRS credentials
const char pin[] = "";  
const char apn[] = ""; 
const char login[] = "";  
const char password[] = "";  

//  OpenCellID API key
const char* apiKey = "pk.*******************************";

// Cell tower info
const char* mcc = "244";
const char* mnc = "91";
const char* lac = "1337";
const char* cid = "23558";
const char* radio = "GSM";

// The GSM SSL client
GSMSSLClient client;
GPRS gprs;
GSM gsmAccess;

void setup() {
  Serial.begin(9600);
  while (!Serial) {} // Wait for the serial connection
MODEM.debug();
  Serial.println("Initializing connection...");

  if (!gsmAccess.begin(pin)) {
    Serial.println("Failed to connect to the GSM network");
    while (true);
  }

  if (gprs.attachGPRS(apn, login, password) != GPRS_READY) {
    Serial.println("Failed to connect to the GPRS network");
    while (true);
  }

  Serial.println("Sending HTTP request...");
  String url = "https://opencellid.org/cell/get?key="+String(apiKey)+"&mcc="+String(mcc)+"&mnc="+String(mnc)+"&lac="+String(lac)+"&cellid="+ String(cid)+"&radio="+String(radio)+"&format=json";
  MODEM.debug();
  if (client.connect("opencellid.org", 443)) {
    Serial.println("Connected to server");
    client.println("GET " + url + "HTTP/1.1");
    client.println("Host: opencellid.org");
    client.println("Connection: close");
    client.println();
  } else {
    Serial.println("Connection failed");
  }

  Serial.println("Waiting for response...");
  String statusLine = client.readStringUntil('\r\n');
  Serial.println("Response: " + statusLine);

  if (statusLine.startsWith("HTTP/1.1 200 OK")) {
    Serial.println("HTTP 200: Success");
    String payload = client.readStringUntil('\n');
    Serial.println("Payload received: " + payload);
    // Additional parsing and actions 
  } else {
    Serial.println("Failed, response status: " + statusLine);
  }

  client.stop();
  Serial.println("Disconnected");
}

void loop() {
  // Empty loop
}`

Serial monitor (Debugging mode activated):

`Initializing connection...
Sending HTTP request...
AT+USECMNG=0,0,"AddTrust_External_CA_Root",1082
>
+USECMNG: 0,0,"AddTrust_External_CA_Root","1d3554048578b03f42424dbf20730a3f"

OK
AT+USECMNG=0,0,"Baltimore_CyberTrust_Root",891
>
+USECMNG: 0,0,"Baltimore_CyberTrust_Root","acb694a59c17e0d791529bb19706a6e4"

OK
AT+USECMNG=0,0,"COMODO_RSA_Certification_Authority",1500
>
+USECMNG: 0,0,"COMODO_RSA_Certification_Authority","1b31b0714036cc143691adc43efdec18"

OK
AT+USECMNG=0,0,"DST_Root_CA_X3",846
>
+USECMNG: 0,0,"DST_Root_CA_X3","410352dc0ff7501b16f0028eba6f45c5"

OK
AT+USECMNG=0,0,"DigiCert_High_Assurance_EV_Root_CA",969
>
+USECMNG: 0,0,"DigiCert_High_Assurance_EV_Root_CA","d474de575c39b2d39c8583c5c065498a"

OK
AT+USECMNG=0,0,"Entrust_Root_Certification_Authority",1173
>
+USECMNG: 0,0,"Entrust_Root_Certification_Authority","d6a5c3ed5ddd3e00c13d87921f1d3fe4"

OK
AT+USECMNG=0,0,"Equifax_Secure_Certificate_Authority",804
>
+USECMNG: 0,0,"Equifax_Secure_Certificate_Authority","67cb9dc013248a829bb2171ed11becd4"

OK
AT+USECMNG=0,0,"GeoTrust_Global_CA",856
>
+USECMNG: 0,0,"GeoTrust_Global_CA","f775ab29fb514eb7775eff053c998ef5"

OK
AT+USECMNG=0,0,"GeoTrust_Primary_Certification_Authority_G3",1026
>
+USECMNG: 0,0,"GeoTrust_Primary_Certification_Authority_G3","b5e83436c910445848706d2e83d4b805"

OK
AT+USECMNG=0,0,"GlobalSign",958
>
+USECMNG: 0,0,"GlobalSign","9414777e3e5efd8f30bd41b0cfe7d030"

OK
AT+USECMNG=0,0,"Go_Daddy_Root_Certificate_Authority_G2",969
>
+USECMNG: 0,0,"Go_Daddy_Root_Certificate_Authority_G2","803abc22c1e6fb8d9b3b274a321b9a01"

OK
AT+USECMNG=0,0,"VeriSign_Class_3_Public_Primary_Certification_Authority_G5",1239
>
+USECMNG: 0,0,"VeriSign_Class_3_Public_Primary_Certification_Authority_G5","cb17e431673ee209fe455793f30afa1c"

OK
AT+USECMNG=2,0,"AmazonRootCA1"

***ERROR***

AT+USECMNG=0,0,"Starfield_Services_Root_Certificate_Authority_G2",1011
>
+USECMNG: 0,0,"Starfield_Services_Root_Certificate_Authority_G2","173574af7b611cebf4f93ce2ee40f9a2"

OK
AT+USOCR=6

+USOCR: 0

OK
AT+USOSEC=0,1,0

OK
AT+USECPRF=0,0,1

OK
AT+USOCO=0,"opencellid.org",443

***ERROR***

+UUSOCL: 0
AT+USOCL=0

***ERROR***
Connection failed
Waiting for response...
Response: 
Failed, response status: 
Disconnected`

pennam, 2024-03-28:
Hi @ShahabYahoo running `openssl s_client -showcerts -connect opencellid.org:443 </dev/null` looks like you will need https://letsencrypt.org/certificates/ to connect to that site. Try to remove `AmazonRootCA1` and add the `ISRG Root X1`

ShahabYahoo, 2024-03-31:
Thank you @pennam, I tried to apply your instructions to my program but sadly faced errors again. I shared the modified program alongside the serial output for debugging. As I am a beginner in Arduino, some functions or commands are a bit difficult and confusing for me to comprehend. So, although I tried to study some tutorials about SSL, certificates, and the module datasheet I still don't have an appropriate perception of what I studied. Indeed this code snippet is a part of my program relevant to my thesis which I've stuck in it. 
These are the steps that I've taken: 
1- I downloaded the "ISRG Root X1 certificate" from this URL:https://letsencrypt.org/certificates/ (Self-signed: der)
2- Then I converted the DER format to C style array through the online tool:https://notisrac.github.io/FileToCArray/
3- Finally, I removed the AmazonRootCA1 certificate and loaded the hex value directly into my program before making an HTTPS request as you can see in the program code

Thank you in advance for taking the time and effort to verify and debug,

Modified program:
`#include <MKRGSM.h>
#include <ArduinoJson.h>

//  GPRS credentials
const char pin[] = "";  
const char apn[] = "internet"; 
const char login[] = "";  
const char password[] = "";  

//  OpenCellID API key
const char* apiKey = "pk.*******************************";

// Cell tower info
const char* mcc = "244";
const char* mnc = "91";
const char* lac = "1337";
const char* cid = "23558";
const char* radio = "GSM";
// ISRG Root X1 Certificate data
const unsigned char isrgrootx1[] PROGMEM = {
  0x30, 0x82, 0x05, 0x6b, 0x30, 0x82, 0x03, 0x53, 0xa0, 0x03, 0x02, 0x01, 0x02, 0x02, 0x11, 0x00, 
0x82, 0x10, 0xcf, 0xb0, 0xd2, 0x40, 0xe3, 0x59, 0x44, 0x63, 0xe0, 0xbb, 0x63, 0x82, 0x8b, 0x00, 
0x30, 0x0d, 0x06, 0x09, 0x2a, 0x86, 0x48, 0x86, 0xf7, 0x0d, 0x01, 0x01, 0x0b, 0x05, 0x00, 0x30, 
0x4f, 0x31, 0x0b, 0x30, 0x09, 0x06, 0x03, 0x55, 0x04, 0x06, 0x13, 0x02, 0x55, 0x53, 0x31, 0x29, 
0x30, 0x27, 0x06, 0x03, 0x55, 0x04, 0x0a, 0x13, 0x20, 0x49, 0x6e, 0x74, 0x65, 0x72, 0x6e, 0x65, 
0x74, 0x20, 0x53, 0x65, 0x63, 0x75, 0x72, 0x69, 0x74, 0x79, 0x20, 0x52, 0x65, 0x73, 0x65, 0x61, 
0x72, 0x63, 0x68, 0x20, 0x47, 0x72, 0x6f, 0x75, 0x70, 0x31, 0x15, 0x30, 0x13, 0x06, 0x03, 0x55, 
0x04, 0x03, 0x13, 0x0c, 0x49, 0x53, 0x52, 0x47, 0x20, 0x52, 0x6f, 0x6f, 0x74, 0x20, 0x58, 0x31, 
0x30, 0x1e, 0x17, 0x0d, 0x31, 0x35, 0x30, 0x36, 0x30, 0x34, 0x31, 0x31, 0x30, 0x34, 0x33, 0x38, 
0x5a, 0x17, 0x0d, 0x33, 0x35, 0x30, 0x36, 0x30, 0x34, 0x31, 0x31, 0x30, 0x34, 0x33, 0x38, 0x5a, 
0x30, 0x4f, 0x31, 0x0b, 0x30, 0x09, 0x06, 0x03, 0x55, 0x04, 0x06, 0x13, 0x02, 0x55, 0x53, 0x31, 
0x29, 0x30, 0x27, 0x06, 0x03, 0x55, 0x04, 0x0a, 0x13, 0x20, 0x49, 0x6e, 0x74, 0x65, 0x72, 0x6e, 
0x65, 0x74, 0x20, 0x53, 0x65, 0x63, 0x75, 0x72, 0x69, 0x74, 0x79, 0x20, 0x52, 0x65, 0x73, 0x65, 
0x61, 0x72, 0x63, 0x68, 0x20, 0x47, 0x72, 0x6f, 0x75, 0x70, 0x31, 0x15, 0x30, 0x13, 0x06, 0x03, 
0x55, 0x04, 0x03, 0x13, 0x0c, 0x49, 0x53, 0x52, 0x47, 0x20, 0x52, 0x6f, 0x6f, 0x74, 0x20, 0x58, 
0x31, 0x30, 0x82, 0x02, 0x22, 0x30, 0x0d, 0x06, 0x09, 0x2a, 0x86, 0x48, 0x86, 0xf7, 0x0d, 0x01, 
0x01, 0x01, 0x05, 0x00, 0x03, 0x82, 0x02, 0x0f, 0x00, 0x30, 0x82, 0x02, 0x0a, 0x02, 0x82, 0x02, 
0x01, 0x00, 0xad, 0xe8, 0x24, 0x73, 0xf4, 0x14, 0x37, 0xf3, 0x9b, 0x9e, 0x2b, 0x57, 0x28, 0x1c, 
0x87, 0xbe, 0xdc, 0xb7, 0xdf, 0x38, 0x90, 0x8c, 0x6e, 0x3c, 0xe6, 0x57, 0xa0, 0x78, 0xf7, 0x75, 
0xc2, 0xa2, 0xfe, 0xf5, 0x6a, 0x6e, 0xf6, 0x00, 0x4f, 0x28, 0xdb, 0xde, 0x68, 0x86, 0x6c, 0x44, 
0x93, 0xb6, 0xb1, 0x63, 0xfd, 0x14, 0x12, 0x6b, 0xbf, 0x1f, 0xd2, 0xea, 0x31, 0x9b, 0x21, 0x7e, 
0xd1, 0x33, 0x3c, 0xba, 0x48, 0xf5, 0xdd, 0x79, 0xdf, 0xb3, 0xb8, 0xff, 0x12, 0xf1, 0x21, 0x9a, 
0x4b, 0xc1, 0x8a, 0x86, 0x71, 0x69, 0x4a, 0x66, 0x66, 0x6c, 0x8f, 0x7e, 0x3c, 0x70, 0xbf, 0xad, 
0x29, 0x22, 0x06, 0xf3, 0xe4, 0xc0, 0xe6, 0x80, 0xae, 0xe2, 0x4b, 0x8f, 0xb7, 0x99, 0x7e, 0x94, 
0x03, 0x9f, 0xd3, 0x47, 0x97, 0x7c, 0x99, 0x48, 0x23, 0x53, 0xe8, 0x38, 0xae, 0x4f, 0x0a, 0x6f, 
0x83, 0x2e, 0xd1, 0x49, 0x57, 0x8c, 0x80, 0x74, 0xb6, 0xda, 0x2f, 0xd0, 0x38, 0x8d, 0x7b, 0x03, 
0x70, 0x21, 0x1b, 0x75, 0xf2, 0x30, 0x3c, 0xfa, 0x8f, 0xae, 0xdd, 0xda, 0x63, 0xab, 0xeb, 0x16, 
0x4f, 0xc2, 0x8e, 0x11, 0x4b, 0x7e, 0xcf, 0x0b, 0xe8, 0xff, 0xb5, 0x77, 0x2e, 0xf4, 0xb2, 0x7b, 
0x4a, 0xe0, 0x4c, 0x12, 0x25, 0x0c, 0x70, 0x8d, 0x03, 0x29, 0xa0, 0xe1, 0x53, 0x24, 0xec, 0x13, 
0xd9, 0xee, 0x19, 0xbf, 0x10, 0xb3, 0x4a, 0x8c, 0x3f, 0x89, 0xa3, 0x61, 0x51, 0xde, 0xac, 0x87, 
0x07, 0x94, 0xf4, 0x63, 0x71, 0xec, 0x2e, 0xe2, 0x6f, 0x5b, 0x98, 0x81, 0xe1, 0x89, 0x5c, 0x34, 
0x79, 0x6c, 0x76, 0xef, 0x3b, 0x90, 0x62, 0x79, 0xe6, 0xdb, 0xa4, 0x9a, 0x2f, 0x26, 0xc5, 0xd0, 
0x10, 0xe1, 0x0e, 0xde, 0xd9, 0x10, 0x8e, 0x16, 0xfb, 0xb7, 0xf7, 0xa8, 0xf7, 0xc7, 0xe5, 0x02, 
0x07, 0x98, 0x8f, 0x36, 0x08, 0x95, 0xe7, 0xe2, 0x37, 0x96, 0x0d, 0x36, 0x75, 0x9e, 0xfb, 0x0e, 
0x72, 0xb1, 0x1d, 0x9b, 0xbc, 0x03, 0xf9, 0x49, 0x05, 0xd8, 0x81, 0xdd, 0x05, 0xb4, 0x2a, 0xd6, 
0x41, 0xe9, 0xac, 0x01, 0x76, 0x95, 0x0a, 0x0f, 0xd8, 0xdf, 0xd5, 0xbd, 0x12, 0x1f, 0x35, 0x2f, 
0x28, 0x17, 0x6c, 0xd2, 0x98, 0xc1, 0xa8, 0x09, 0x64, 0x77, 0x6e, 0x47, 0x37, 0xba, 0xce, 0xac, 
0x59, 0x5e, 0x68, 0x9d, 0x7f, 0x72, 0xd6, 0x89, 0xc5, 0x06, 0x41, 0x29, 0x3e, 0x59, 0x3e, 0xdd, 
0x26, 0xf5, 0x24, 0xc9, 0x11, 0xa7, 0x5a, 0xa3, 0x4c, 0x40, 0x1f, 0x46, 0xa1, 0x99, 0xb5, 0xa7, 
0x3a, 0x51, 0x6e, 0x86, 0x3b, 0x9e, 0x7d, 0x72, 0xa7, 0x12, 0x05, 0x78, 0x59, 0xed, 0x3e, 0x51, 
0x78, 0x15, 0x0b, 0x03, 0x8f, 0x8d, 0xd0, 0x2f, 0x05, 0xb2, 0x3e, 0x7b, 0x4a, 0x1c, 0x4b, 0x73, 
[... 220 lines cut ...]

OK
AT+USOCR=6

+USOCR: 0

OK
AT+USOSEC=0,1,0

OK
AT+USECPRF=0,0,1

OK   AT+USOCO=0,"opencellid.org",443

_**ERROR**_

+UUSOCL: 0
AT+USOCL=0

_**ERROR**_
Connection failed
Waiting for response...
Response: 
Failed, response status: 
Disconnected`

ShahabYahoo, 2024-04-02:
Dear @pennam I tested the "MKR GSM SSL Web Client" example from the official Arduino website, and it does not work either as you can see in the code and output below. Do you have any resolution?

`// libraries
#include <MKRGSM.h>

#include "arduino_secrets.h"
// Please enter your sensitive data in the Secret tab or arduino_secrets.h
// PIN Number

const char PINNUMBER[]     = SECRET_PINNUMBER;
// APN data

const char GPRS_APN[]      = SECRET_GPRS_APN;

const char GPRS_LOGIN[]    = SECRET_GPRS_LOGIN;

const char GPRS_PASSWORD[] = SECRET_GPRS_PASSWORD;

// initialize the library instance

GSMSSLClient client;

GPRS gprs;

GSM gsmAccess;

// URL, path and port (for example: arduino.tips)
char server[] = "arduino.tips";
char path[] = "/asciilogo.txt";
int port = 443; // port 443 is the default for HTTPS

void setup() {

  // initialize serial communications and wait for port to open:

  Serial.begin(9600);

  while (!Serial) {

    ; // wait for serial port to connect. Needed for native USB port only

  }
MODEM.debug();
  Serial.println("Starting Arduino web client.");

  // connection state

  bool connected = false;

  // After starting the modem with GSM.begin()

  // attach the shield to the GPRS network with the APN, login and password

  while (!connected) {

    if ((gsmAccess.begin(PINNUMBER) == GSM_READY) &&

        (gprs.attachGPRS(GPRS_APN, GPRS_LOGIN, GPRS_PASSWORD) == GPRS_READY)) {

      connected = true;

    } else {

      Serial.println("Not connected");

      delay(1000);

    }

  }

  Serial.println("connecting...");

  // if you get a connection, report back via serial:

  if (client.connect(server, port)) {

    Serial.println("connected");

    // Make a HTTP request:

    client.print("GET ");

    client.print(path);

    client.println(" HTTP/1.1");

    client.print("Host: ");

    client.println(server);

    client.println("Connection: close");

    client.println();

  } else {

    // if you didn't get a connection to the server:

    Serial.println("connection failed");

  }
}

void loop() {

  // if there are incoming bytes available

  // from the server, read them and print them:

  if (client.available()) {

    char c = client.read();

    Serial.print(c);

  }

  // if the server's disconnected, stop the client:

  if (!client.available() && !client.connected()) {

    Serial.println();

    Serial.println("disconnecting.");

    client.stop();

    // do nothing forevermore:

    for (;;)

      ;

  }
}`

**Serial Monitor:**

`Starting Arduino web client.
AT

OK
AT+IPR=921600

OK
AT

OK
AT+UPSV=3

OK
AT+CPIN?

ERROR
AT+CPIN?

ERROR
AT+CPIN?

+CPIN: SIM PIN

OK
AT+CPIN="5271"

OK
AT+CMGF=1

OK
AT+UDCONF=1,1

OK
AT+CTZU=1

OK
AT+UDTMFD=1,2

OK
AT+CREG?

+CREG: 0,0

OK
AT+CREG?

+CREG: 0,0

OK
AT+CREG?

+CREG: 0,0

OK
AT+CREG?

+CREG: 0,0
OK
AT+CREG?

+CREG: 0,0

OK
AT+CREG?

+CREG: 0,0

OK
AT+CREG?

+CREG: 0,0

OK
AT+CREG?

+CREG: 0,0

OK
AT+CREG?

+CREG: 0,0

OK
AT+CREG?

+CREG: 0,0

OK
AT+CREG?

+CREG: 0,0

OK

+UMWI: 0,1

+UMWI: 0,2

+UMWI: 0,3

+UMWI: 0,4
AT+CREG?

+CREG: 0,0
OK
AT+CREG?

+CREG: 0,0

OK
AT+CREG?

+CREG: 0,0

OK
AT+CREG?

+CREG: 0,0

OK
AT+CREG?

+CREG: 0,0

OK
AT+CREG?

+CREG: 0,0

OK
AT+CREG?

+CREG: 0,0

OK
AT+CREG?

+CREG: 0,0

OK
AT+CREG?

+CREG: 0,0

OK
AT+CREG?

+CREG: 0,0

OK
AT+CREG?

+CREG: 0,0

OK
AT+CREG?

+CREG: 0,0

OK
AT+CREG?

+CREG: 0,0

OK
AT+CREG?

+CREG: 0,1

OK
AT+UCALLSTAT=1

OK
AT+CGATT=1

OK
AT+UPSD=0,1,"internet.saunalahti"

OK
AT+UPSD=0,6,3

OK
AT+UPSD=0,2,""

OK
AT+UPSD=0,3,""

OK
AT+UPSD=0,7,"0.0.0.0"

OK
AT+UPSDA=0,3

OK
AT+UPSND=0,8

+UPSND: 0,8,1

OK
connecting...
AT+USECMNG=0,0,"AddTrust_External_CA_Root",1082
>
+USECMNG: 0,0,"AddTrust_External_CA_Root","1d3554048578b03f42424dbf20730a3f"
OK
AT+USECMNG=0,0,"Baltimore_CyberTrust_Root",891
>
+USECMNG: 0,0,"Baltimore_CyberTrust_Root","acb694a59c17e0d791529bb19706a6e4"

OK
AT+USECMNG=0,0,"COMODO_RSA_Certification_Authority",1500
>
+USECMNG: 0,0,"COMODO_RSA_Certification_Authority","1b31b0714036cc143691adc43efdec18"

OK
AT+USECMNG=0,0,"DST_Root_CA_X3",846
>
+USECMNG: 0,0,"DST_Root_CA_X3","410352dc0ff7501b16f0028eba6f45c5"

OK
AT+USECMNG=0,0,"DigiCert_High_Assurance_EV_Root_CA",969
>
+USECMNG: 0,0,"DigiCert_High_Assurance_EV_Root_CA","d474de575c39b2d39c8583c5c065498a"

OK
AT+USECMNG=0,0,"Entrust_Root_Certification_Authority",1173
>
+USECMNG: 0,0,"Entrust_Root_Certification_Authority","d6a5c3ed5ddd3e00c13d87921f1d3fe4"

OK
AT+USECMNG=0,0,"Equifax_Secure_Certificate_Authority",804
>
+USECMNG: 0,0,"Equifax_Secure_Certificate_Authority","67cb9dc013248a829bb2171ed11becd4"

OK
AT+USECMNG=0,0,"GeoTrust_Global_CA",856
>
+USECMNG: 0,0,"GeoTrust_Global_CA","f775ab29fb514eb7775eff053c998ef5"

OK
AT+USECMNG=0,0,"GeoTrust_Primary_Certification_Authority_G3",1026
>
+USECMNG: 0,0,"GeoTrust_Primary_Certification_Authority_G3","b5e83436c910445848706d2e83d4b805"

OK
AT+USECMNG=0,0,"GlobalSign",958
>
+USECMNG: 0,0,"GlobalSign","9414777e3e5efd8f30bd41b0cfe7d030"

OK
AT+USECMNG=0,0,"Go_Daddy_Root_Certificate_Authority_G2",969
>
+USECMNG: 0,0,"Go_Daddy_Root_Certificate_Authority_G2","803abc22c1e6fb8d9b3b274a321b9a01"
OK
AT+USECMNG=0,0,"VeriSign_Class_3_Public_Primary_Certification_Authority_G5",1239
>
+USECMNG: 0,0,"VeriSign_Class_3_Public_Primary_Certification_Authority_G5","cb17e431673ee209fe455793f30afa1c"

OK
AT+USECMNG=2,0,"AmazonRootCA1"

ERROR
AT+USECMNG=0,0,"Starfield_Services_Root_Certificate_Authority_G2",1011
>
+USECMNG: 0,0,"Starfield_Services_Root_Certificate_Authority_G2","173574af7b611cebf4f93ce2ee40f9a2"

OK
AT+USOCR=6

+USOCR: 0

OK
AT+USOSEC=0,1,0

OK
AT+USECPRF=0,0,1

OK
AT+USOCO=0,"arduino.tips",443

ERROR

+UUSOCL: 0
AT+USOCL=0

ERROR
connection failed

disconnecting.`